MatthewPierson/1033-OTA-Downgrader #1: failure on restore.sh

gb160, 2019-10-11:
`**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Please enter device ID (iPhone6,1 or iPhone6,2 only) ONLY THE iPHONE 5s IS SUPPORTED
iPhone6,2
[Log] Vaild device, continuing
[Log] Getting current APNonce
227840674916
47080312e71eda7ccd8eec33af095f41710fe8db
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "227840674916" to dec:227840674916 hex:350c5ccc64
[TSSC] manually specified ApNonce to use, parsed "47080312e71eda7ccd8eec33af095f41710fe8db" to hex:47080312e71eda7ccd8eec33af095f41710fe8db
[TSSC] opening restore/BuildManifest_iPhone6,2_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPhone6,2 IS being signed!
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
n
[Log] SHSH saved
[Log] Starting restore process

[Log] Copying SEP and Baseband
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/OTA.shsh is done
Found device iPhone6,2 n53ap
[TSSC] opening restore/BuildManifest_iPhone6,2_1033_OTA.plist
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Did set SEP+baseband path and firmware
[WARNING] Failed to read BasebandGoldCertID from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandGoldCertID. This might result in invalid baseband signing status information
[WARNING] Failed to read BasebandSerialNumber from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandSerialNumber size. This might result in invalid baseband signing status information
[TSSC] opening restore/BuildManifest_iPhone6,2_1033_OTA.plist
[TSSR] User specified to request only a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Found device in DFU mode
[Error] unsupported device mode, please put device in recovery mode or normal mode
[Error] Fail code=-3
Failed with errorcode=-3
[Log] Futurerestoring complete

**************** Downgrade complete! Enjoy 10.3.3 =) ****************
**************** Follow me on twitter @mosk_i for help/updates ******
`

MatthewPierson, 2019-10-12:
This is an issue some people get on Catalina. Don't know why yet but I recommend using Mojave if Catalina doesn't work for you sorry.

motoissa, 2019-10-12:
I'm running Mojave 
**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Please enter device ID (iPhone6,1 or iPhone6,2 only) ONLY THE iPHONE 5s IS SUPPORTED
iPhone6,1
[Log] Vaild device, continuing
[Log] Getting current APNonce
5550335620560
26b40fe2298c628f1a188945ef2dff3d918fbaf8
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "5550335620560" to dec:5550335620560 hex:50c49c875d0
[TSSC] manually specified ApNonce to use, parsed "26b40fe2298c628f1a188945ef2dff3d918fbaf8" to hex:26b40fe2298c628f1a188945ef2dff3d918fbaf8
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPhone6,1 IS being signed!
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
n
[Log] SHSH saved
[Log] Starting restore process

[Log] Copying SEP and Baseband
cp: 10.3.3.custom/Firmware/Mav7Mav8-7.60.00.Release.bbfw: No such file or directory
cp: 10.3.3.custom/Firmware/all_flash/sep-firmware.n53.RELEASE.im4p: No such file or directory
cp: 10.3.3.custom/Firmware/all_flash/sep-firmware.n51.RELEASE.im4p: No such file or directory
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/OTA.shsh is done
Found device iPhone6,1 n51ap
[Error] failed to read SEP
[Error] Fail code=-15
Failed with errorcode=-15
[Log] Futurerestoring complete

MatthewPierson, 2019-10-12:
You need to run ./prep.sh again and make sure that the sep/baseband files are in  10.3.3.custom/Firmware for baseband and 10.3.3.custom/Firmware/all_flash/ for sep after the script has run.

motoissa, 2019-10-12:
**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Please enter device ID (iPhone6,1 or iPhone6,2 only) ONLY THE iPHONE 5s IS SUPPORTED
iPhone6,1
[Log] Vaild device, continuing
[Log] Getting current APNonce
5550335620560
7c8ae5776108260e626a8c7381b96bc4815652b6
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "5550335620560" to dec:5550335620560 hex:50c49c875d0
[TSSC] manually specified ApNonce to use, parsed "7c8ae5776108260e626a8c7381b96bc4815652b6" to hex:7c8ae5776108260e626a8c7381b96bc4815652b6
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPhone6,1 IS being signed!
[Log] Starting restore process

[Log] Copying SEP and Baseband
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/OTA.shsh is done
Found device iPhone6,1 n51ap
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Did set SEP+baseband path and firmware
[WARNING] Failed to read BasebandGoldCertID from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandGoldCertID. This might result in invalid baseband signing status information
[WARNING] Failed to read BasebandSerialNumber from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandSerialNumber size. This might result in invalid baseband signing status information
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] User specified to request only a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Found device in DFU mode
[Error] unsupported device mode, please put device in recovery mode or normal mode
[Error] Fail code=-3
Failed with errorcode=-3
[Log] Futurerestoring complete

**************** Downgrade complete! Enjoy 10.3.3 =) ****************
**************** Follow me on twitter @mosk_i for help/updates ******

motoissa, 2019-10-12:
**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Please enter device ID (iPhone6,1 or iPhone6,2 only) ONLY THE iPHONE 5s IS SUPPORTED
iPhone6,1
[Log] Vaild device, continuing
[Log] Getting current APNonce
5550335620560
3a88b7c3802f2f0510abc432104a15ebd8bd7154
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "5550335620560" to dec:5550335620560 hex:50c49c875d0
[TSSC] manually specified ApNonce to use, parsed "3a88b7c3802f2f0510abc432104a15ebd8bd7154" to hex:3a88b7c3802f2f0510abc432104a15ebd8bd7154
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPhone6,1 IS being signed!
[Log] Starting restore process

[Log] Copying SEP and Baseband
cp: 10.3.3.custom/Firmware/Mav7Mav8-7.60.00.Release.bbfw: No such file or directory
cp: 10.3.3.custom/Firmware/all_flash/sep-firmware.n53.RELEASE.im4p: No such file or directory
cp: 10.3.3.custom/Firmware/all_flash/sep-firmware.n51.RELEASE.im4p: No such file or directory
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
INFO: device serial number is F17M4W8TFF9V
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/OTA.shsh is done
Found device iPhone6,1 n51ap
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Did set SEP+baseband path and firmware
[WARNING] Failed to read BasebandGoldCertID from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandGoldCertID. This might result in invalid baseband signing status information
[WARNING] Failed to read BasebandSerialNumber from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandSerialNumber size. This might result in invalid baseband signing status information
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] User specified to request only a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Found device in Recovery mode
Device already in Recovery mode
Found device in Recovery mode
Identified device as n51ap, iPhone6,1
Extracting BuildManifest from iPSW
Product version: 10.3.3
Product build: 14G60 Major: 14
Device supports IMG4: true
Got ApNonce from device: 3a 88 b7 c3 80 2f 2f 05 10 ab c4 32 10 4a 15 eb d8 bd 71 54 
checking APTicket to be valid for this restore...
Verified ECID in APTicket matches device ECID
checking APTicket to be valid for this restore...
Verified ECID in APTicket matches device ECID
[Error] im4m_buildidentity_check_cb: can't find any identity which matches all hashes inside IM4M
[Error] getBuildIdentityForIM4M: found buildidentity, but can't read information
BuildIdentity selected for restore:
BuildNumber : 14G60
BuildTrain : Greensburg
DeviceClass : n51ap
FDRSupport : NO
RestoreBehavior : Erase
Variant : Customer Erase Install (IPSW)

BuildIdentiy valid for the APTicket:
IM4M isn't valid for any restore with this BuildManifest
This APTicket can't be used for restoring this firmware
IM4M isn't valid for any restore with this BuildManifest
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Extracting filesystem from iPSW
[==================================================] 100.0%
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (653333 bytes)...
waiting for device to reconnect... 
Getting SepNonce in recovery mode... 2f 18 76 de a7 e1 f0 cb 77 46 af c6 28 36 e4 e9 2a 0a ce 1e 
Getting ApNonce in recovery mode... 3a 88 b7 c3 80 2f 2f 05 10 ab c4 32 10 4a 15 eb d8 bd 71 54 
[WARNING] Setting bgcolor to green! If you don't see a green screen, then your device didn't boot iBEC correctly
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (11640 bytes)...
ramdisk-size=0x10000000
Extracting 058-74940-063.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41583638 bytes)...
Extracting DeviceTree.n51ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (107919 bytes)...
Extracting kernelcache.release.iphone6...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12178427 bytes)...
Trying to fetch new signing tickets
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received signing tickets
About to restore device... 
Waiting for device...
it never does anything

gb160, 2019-10-12:
> This is an issue some people get on Catalina. Don't know why yet but I recommend using Mojave if Catalina doesn't work for you sorry.

You were right about irecovery, I've run the steps in the script manually and everything works perfectly until sending the ibss/ibec files...its always the ibec file that it gets stuck at, and times out eventually, obviously the next stages of the script won't work correctly after that.

gb160, 2019-10-12:
@MatthewPierson do u think that the Catalina issues could be related to the switch to using zsh as the default shell?

gb160, 2019-10-12:
After installing a fresh install on Mojave, I'm getting the same issue....fails to correctly send the ibss/ibec files:

`**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************

Plug device into computer in DFU Mode

THIS WILL FAIL UP TO 40 TIMES, THIS IS NORMAL JUST RE-RUN THIS SCRIPT

There is no current way around this, it's just the nature of the exploit on the 5s

Thanks to Merc (@Vyce_Merculous) for helping me fix stuff with this script!

... Waiting 10 seconds for you to actually read this before continuing ...
[Log] Removing old files

[Log] Entering PWNDFU Mode
No matching processes belonging to you were found
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:000000350C5CCC64 IBFL:1C SRTG:[iBoot-1704.10]
Device is now in pwned DFU Mode.
(12.66 seconds)
2019-10-12 12:45:20.389 system_profiler[527:7178] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 12:45:20.389 system_profiler[527:7178] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 12:45:20.390 system_profiler[527:7178] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 12:45:20.390 system_profiler[527:7178] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 12:45:20.392 system_profiler[527:7178] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 12:45:20.392 system_profiler[527:7178] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
1
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:000000350C5CCC64 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Successfully applied patches
Resetting device state
* This will effectiveley disable pwned DFU Mode
* Only the signature patches will remain
Device is now ready to accept unsigned images
[Log] Device is in PWNDFU Mode with Sigchcks removed

[Log] Putting device into PWNDREC mode
[==================================================] 100.0%
ERROR: Unable to connect to device
[Log] Device is now in PWNDREC mode

**************** PWNing Completed. Please run restore.sh ****************
`

gb160, 2019-10-12:
On iPad Air, it doesn't get that far before it errors out:

`**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************

Plug device into computer in DFU Mode

THIS WILL FAIL UP TO 40 TIMES, THIS IS NORMAL JUST RE-RUN THIS SCRIPT

There is no current way around this, it's just the nature of the exploit on the 5s

Thanks to Merc (@Vyce_Merculous) for helping me fix stuff with this script!

... Waiting 10 seconds for you to actually read this before continuing ...
[Log] Removing old files

[Log] Entering PWNDFU Mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:10 ECID:000004B7CB02F4D0 IBFL:1C SRTG:[iBoot-1704.10]
Device is now in pwned DFU Mode.
(12.55 seconds)
2019-10-12 13:14:41.894 system_profiler[28586:92346] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 13:14:41.895 system_profiler[28586:92346] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 13:14:41.895 system_profiler[28586:92346] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 13:14:41.896 system_profiler[28586:92346] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 13:14:41.897 system_profiler[28586:92346] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-12 13:14:41.898 system_profiler[28586:92346] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
1
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:10 ECID:000004B7CB02F4D0 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Successfully applied patches
Resetting device state
* This will effectiveley disable pwned DFU Mode
* Only the signature patches will remain
Traceback (most recent call last):
  File "rmsigchks.py", line 106, in <module>
    main()
  File "rmsigchks.py", line 99, in main
    device.ctrl_transfer(HOST2DEVICE, DFU_ABORT, 0, 0, 0, 0)
  File "/Users/gavin/Desktop/iPad_1033-OTA-Downgrader-master/ipwndfu_public/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/Users/gavin/Desktop/iPad_1033-OTA-Downgrader-master/ipwndfu_public/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/Users/gavin/Desktop/iPad_1033-OTA-Downgrader-master/ipwndfu_public/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
[Log] Device is in PWNDFU Mode with Sigchcks removed

[Log] Putting device into PWNDREC mode
[Log] Device is now in PWNDREC mode

**************** PWNing Completed. Please run restore.sh ****************
`

gb160, 2019-10-13:
@MatthewPierson Upon some further investigation using a linux machine, irecovery also fails to send ibec.final, ibss.final sends fine....exactly the same behaviour as on my MacBook....gets stuck at 80.9% every time.

I can only conclude that something's not right with ibss.final

MatthewPierson, 2019-10-13:
iBEC stops sending half way if the device was not put into PWNREC mode properly or if iBSS had an issue (even if it reported as sending fully). My patches are fine and have been used successfully by many people. You can try manually sending iBSS/iBEC after entering PWNDFU mode and removing sigchecks with "pythong rmsigchks.py" to see if that fixes the issue.

gb160, 2019-10-13:
PROGRESS!!! Kind of lol.

I was just about to pack it in, but then thought id try my iPad Air and to send the ibss/ibec files from my linux machine (raspberry pi)...so I ssh'd the files over to the pi, put the iPad in pwndfu state, and ran rmsigchks.py from my mac...this seemed to work so I plugged the iPad into the pi , and used irecovery to SUCCESSFULLY send ibss/ibec to the iPad, worked first time.

then I plugged the iPad back into the Mac, ran restore.sh, and lo and behold my iPad is now running 10.3.3.

Im well happy with it mate, it was unusable on ios12.

Just need to work out what the hell is going on with the 5s now.

gb160, 2019-10-13:
More progress, 5S is now up and running 10.3.3

I had to resort to using my Pi again for sending the ibss/ibec again, but it worked after a couple of tries which allowed me to downgrade successfully at long last.

Apologies for doubting your patch method mate, I would conclude that the issue I was having must be down to the hardware (latest MacBook Pro) in combination with irecovery...they just don't play nice together.
As this mac only has USB C ports, I'm forced to use a USB adapter so that's another piece of hardware in the way that could've been causing the issue.

Still worth remembering if anyone else comes across this nasty little issue, which I suppose someone will at some point, especially if you start adding more devices to this.
Wasn't too difficult to install the dependencies on the pi, took about 10 minutes and I wish I'd tried it sooner.

Cheers buddy!

closed 2019-10-13 by gb160

juul0003v, 2022-03-06:
should i open Mav7Mav8.release.bffw i have it pasted approx 22.8MB in notes on 15.3.1