MatthewPierson/1033-OTA-Downgrader #10: Error on High Sierra

elcomix97edu, 2019-10-12:
I'm trying to run the script in High Sierra and all work but restore.sh give me this error:Reason: Incompatible library version: futurerestore requires version 6.0.0 or later, but libusbmuxd.4.dylib provides version 5.0.0

I need update my mac to Mojave or there is a way to execute the script in High Sierra

MatthewPierson, 2019-10-12:
There is no High Sierra support as I can't test it due to only having Mojave machines.

nikosntoykas, 2019-10-12:
This is very very sad. Gοοdbye 5s...... if Sierra is not supported why I'm losing my day today? Please write it to the description. THERE IS NO SIERRA SUPPORT.

Merculous, 2019-10-12:
I'm working on a fix. It will fix most of the issues, if not all.

mittovipi, 2019-10-13:
High sierra works fine

mittovipi, 2019-10-13:
it needs a little modification to work on High Sierra

consolelogluke, 2019-10-13:
What's the modification?

nikosntoykas, 2019-10-13:
> it needs a little modification to work on High Sierra

oh yes;!

mittovipi, 2019-10-13:
Work on High Sierra
http://bit.ly/highsierradowngrade

mittovipi, 2019-10-13:
I am just a user. I am not a dev

elcomix97edu, 2019-10-13:
> Work on High Sierra
> http://bit.ly/highsierradowngrade

I try your version and futurerestore load and inject iBEC correctly but now i have this errors:
ERROR: Unable to find required ApNonce in parameters
ERROR: Unable to add img4 tags to TSS request
[Error] ERROR: Unable to get signing tickets for SEP

consolelogluke, 2019-10-13:
I’m getting the same issue as @elcomix97edu

mittovipi, 2019-10-13:
move the ota.json file to / tmp on macos

mittovipi, 2019-10-13:
Please try again
http://bit.ly/highsierradowngrade5s

MatthewPierson, 2019-10-13:
@mittovipi What have you actually changed besides the language of the logs and the names of the scripts? After looking through them I can't find any changes at all that would indicate a fix for High Sierra machines. I would appreciate you not distributing edited copies of my scripts that claim to have changes that allow for more OS support when in reality they don't.

consolelogluke, 2019-10-13:
@MatthewPierson It actually fixes the issue with High Sierra though, I got way further than I could before. In a minute I’ll try to find the changes.

consolelogluke, 2019-10-13:
@mittovipi The same error happens with the second version.

MatthewPierson, 2019-10-13:
Any changes you had with his “edits” are pure placebo. Check the files, there is no functional change to enable High Sierra support.

mittovipi, 2019-10-13:
sorry.  I will delete all of them

consolelogluke, 2019-10-13:
Really? I'll try again without @mittovipi's changes and see if I get as far as I did *with* his edits.

mittovipi, 2019-10-13:
@MatthewPierson  I do not know. but it actually works on High Sierra
https://youtu.be/YH0actPDdiY

consolelogluke, 2019-10-13:
Yeah, it got way further when using @mittovipi's version than when using the base version.

MatthewPierson, 2019-10-13:
My stuff may work on high Sierra I just don’t support it as I cannot test it. As I’ve said twice already, he has not made any functional changes to the script. When you say it got way further what do you specifically mean? If you have changes that add proper support for High Sierra please submit a pull request and I will review the changes. But unless you do that don’t link out to edited copies with no functional changes.

consolelogluke, 2019-10-13:
He must have changed something because before, futurerestore wouldn't even start but with his changes, the custom iBEC is injected and I get as far as the screen turning green for a second or so.

MatthewPierson, 2019-10-13:
How many times do I need to say it? Open the scripts and compare them! The file names are different, the language in the logs are different and that is it. If I have missed something during my review of the files then please inform me on what that is and I will apologize and take everything I said back.

consolelogluke, 2019-10-13:
I no longer have the modified version and it can’t be downloaded anymore but when I had it I noticed some changes in other files, not the scripts.

MatthewPierson, 2019-10-16:
Closing as nothing is happening here anymore and it got very off topic

closed 2019-10-16 by MatthewPierson