MatthewPierson/1033-OTA-Downgrader #3: got a problem when execute "./prep.sh"

mgfjxxiexiaolong, 2019-10-12:
[image]

MatthewPierson, 2019-10-12:
Has already been fixed in latest version. Please update and try again!

mgfjxxiexiaolong, 2019-10-12:
got a problem when execute "./pwn.sh"
[image]

gb160, 2019-10-12:
> got a problem when execute "./pwn.sh"
> [image]
> [image]

Looks like your phone isn't in DFU mode.

MatthewPierson, 2019-10-12:
Just keep trying. Success rate is low as noted in the script

mgfjxxiexiaolong, 2019-10-12:
I got this right now. It seems failed. I still on ios12.
[image]

delterous, 2019-10-12:
Im trying to use it on an iPad mini 2 and still failing at pwn.sh

success rate must be really bad

mgfjxxiexiaolong, 2019-10-12:
> Im trying to use it on an iPad mini 2 and still failing at pwn.sh
> 
> success rate must be really bad

I failed in this step because of entering wrong mode, the phone is black on screen in DFU mode, make sure that.

gb160, 2019-10-12:
> I got this right now. It seems failed. I still on ios12.
> [image]

Catalina right?
That looks like something’s gone wrong in the install.sh.

Starbix, 2019-10-12:
@mgfjxxiexiaolong Like @gb160 said that's because of an error in `install.sh`. `install.sh` tries to create a folder `/rsu` which fails in Catalina because it now uses a read-only system volume.
You need to disable SIP and then remount the system volume with read/write like this: `sudo mount -uw /`

stratm0r, 2019-10-12:
Will it work with OS X Mavericks?

mgfjxxiexiaolong, 2019-10-12:
> @mgfjxxiexiaolong Like @gb160 said that's because of an error in `install.sh`. `install.sh` tries to create a folder `/rsu` which fails in Catalina because it now uses a read-only system volume.
> You need to disable SIP and then remount the system volume with read/write like this: `sudo mount -uw /`

Im on Mojave....
[image]

Starbix, 2019-10-12:
> Im on Mojave....
> [image]

Is the /rsu directory present?

MatthewPierson, 2019-10-12:
@mgfjxxiexiaolong I would create a 35GB ish partition and install macos Mojave to it. Then run all the scripts there

mgfjxxiexiaolong, 2019-10-12:
> > Im on Mojave....
> > [image]
> 
> Is the /rsu directory present?

Of course.
[image]

mgfjxxiexiaolong, 2019-10-12:
> @mgfjxxiexiaolong I would create a 35GB ish partition and install macos Mojave to it. Then run all the scripts there

OK, Wait for goods news.

liuhouer, 2019-10-13:
[image]

mojave 10.14.6

cuiyihaohao, 2019-10-13:
> [image]
> 
> mojave 10.14.6 with ipad mini2 DFU mode connect my mac

在这里键入ipad或者iPhone，他问你是ipad或者是iPhone而不是问你是否降级。

MatthewPierson, 2019-10-13:
That error is fixed a few commits ago. Please open new issues instead of adding to old ones @liuhouer

liuhouer, 2019-10-13:
> > [image]
> > mojave 10.14.6 with ipad mini2 DFU mode connect my mac
> 
> 在这里键入ipad或者iPhone，他问你是ipad或者是iPhone而不是问你是否降级。

Thanks bro... I try again

mgfjxxiexiaolong, 2019-10-16:
[https://github.com/MatthewPierson/1033-OTA-Downgrader/issues/18](url)

closed 2019-10-16 by mgfjxxiexiaolong