MatthewPierson/1033-OTA-Downgrader #5: getting this trying to run "./prep.sh"  it on a iPhone 5s // iPhone6,2 - bspatch: bin/futurerestore: No such file or directory

cmonzz, 2019-10-12:
getting this trying to run it on a iPhone 5s // iPhone6,2
running this is in High Sieera

!  ./prep.sh 

****** error ***** - this error cuase ./restore error too.

bspatch: bin/futurerestore: No such file or directory
mv: rename tmp/futurerestore_new to bin/futurerestore: No such file or directory
bspatch: bin/futurerestore: No such file or directory
mv: rename tmp/futurerestore_final to bin/futurerestore: No such file or directory
[Log] Making sure permissions are set for binaries

*******************

**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing files from previous runs
[Log] Downloading programs from S0uthwest's Github releases
[Log] Downloading igetnonce
unzip:  cannot find or open igetnonce_macOS_v14.zip, igetnonce_macOS_v14.zip.zip or igetnonce_macOS_v14.zip.ZIP.
mv: rename igetnonce to bin/igetnonce: No such file or directory
[Log] Done igetnonce
[Log] Downloading tsschecker
unzip:  cannot find or open tsschecker_macOS_v355.zip, tsschecker_macOS_v355.zip.zip or tsschecker_macOS_v355.zip.ZIP.
mv: rename tsschecker to bin/tsschecker: No such file or directory
[Log] Done tsschecker
[Log] Downloading futurerestore
unzip:  cannot find or open futurerestore_macOS_v245.zip, futurerestore_macOS_v245.zip.zip or futurerestore_macOS_v245.zip.ZIP.
mv: rename futurerestore to bin/futurerestore: No such file or directory
[Log] Done fututrerestore
[Log] Are you downgrading an iPhone or an iPad?
iphone
[Log] What model iPhone are you using? E.G iPhone6,1 or iPhone6,2
iPhone6,2
[Log] Supported model!
[Log] Unzipping IPSW
Archive:  iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/058-74917-062.dmg
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/._058-74917-062.dmg
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/058-74940-063.dmg
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/058-75381-062.dmg
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/BuildManifest.plist
   creating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/
   creating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/applelogo@2x~iphone.s5l8960x.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/applelogo@2x~iphone.s5l8960x.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/applelogo@2x~iphone.s8000.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/applelogo@2x~iphone.s8000.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging0@2x~iphone.s5l8960x.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging0@2x~iphone.s5l8960x.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging0@2x~iphone.s8000.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging0@2x~iphone.s8000.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging0@2x~iphone.s8003.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging0@2x~iphone.s8003.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging1@2x~iphone.s5l8960x.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging1@2x~iphone.s5l8960x.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging1@2x~iphone.s8000.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging1@2x~iphone.s8000.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging1@2x~iphone.s8003.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterycharging1@2x~iphone.s8003.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batteryfull@2x~iphone.s5l8960x.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batteryfull@2x~iphone.s5l8960x.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batteryfull@2x~iphone.s8003.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batteryfull@2x~iphone.s8003.im4p.plist
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterylow0@2x~iphone.s5l8960x.im4p
  inflating: iPhone_4.0_64bit_10.3.3_14G60_Restore/Firmware/all_flash/batterylow0@2x~iphone.s5l8960x.im4p.plist
[... 169 lines cut ...]
  adding: Firmware/dfu/iBEC.iphone6.RELEASE.im4p (stored 0%)
  adding: Firmware/dfu/iBEC.iphone6.RELEASE.im4p.plist (stored 0%)
  adding: Firmware/dfu/iBEC.n69.RELEASE.im4p (stored 0%)
  adding: Firmware/dfu/iBEC.n69.RELEASE.im4p.plist (stored 0%)
  adding: Firmware/dfu/iBEC.n69u.RELEASE.im4p (stored 0%)
  adding: Firmware/dfu/iBEC.n69u.RELEASE.im4p.plist (stored 0%)
  adding: Firmware/dfu/iBSS.iphone6.RELEASE.im4p (stored 0%)
  adding: Firmware/dfu/iBSS.iphone6.RELEASE.im4p.plist (stored 0%)
  adding: Firmware/dfu/iBSS.n69.RELEASE.im4p (stored 0%)
  adding: Firmware/dfu/iBSS.n69.RELEASE.im4p.plist (stored 0%)
  adding: Firmware/dfu/iBSS.n69u.RELEASE.im4p (stored 0%)
  adding: Firmware/dfu/iBSS.n69u.RELEASE.im4p.plist (stored 0%)
  adding: Firmware/Mav10-5.62.00.Release.bbfw (stored 0%)
  adding: Firmware/Mav10-5.62.00.Release.plist (stored 0%)
  adding: Firmware/Mav7Mav8-7.60.00.Release.bbfw (stored 0%)
  adding: Firmware/Mav7Mav8-7.60.00.Release.plist (stored 0%)
  adding: Firmware/usr/ (stored 0%)
  adding: Firmware/usr/local/ (stored 0%)
  adding: Firmware/usr/local/standalone/ (stored 0%)
  adding: Restore.plist (stored 0%)
  adding: kernelcache.release.iphone6 (stored 0%)
  adding: kernelcache.release.iphone8b (stored 0%)
[Log] Preperations complete
**************** Modified IPSW created successfully ****************
**************** Prep Completed. Please run pwn.sh ****************

MatthewPierson, 2019-10-12:
Did you run install.sh ?

cmonzz, 2019-10-12:
Yes, already done.

cmonzz, 2019-10-12:
it is working now but have fresh installed Mojave instead. Thank you so much

closed 2019-10-12 by cmonzz