MatthewPierson/1033-OTA-Downgrader #6: It can't work on macos catalina

kaaaaaav, 2019-10-12:
macos catalina doesn't work at pwn.sh

consolelogluke, 2019-10-12:
What error are you getting?

MatthewPierson, 2019-10-13:
Please give more details on what is not working/errors you are getting or I cannot help you.

Richard-19, 2019-11-12:
Waiting 10 seconds to allow you to enter DFU mode。And it doesn't work

Richard-19, 2019-11-12:
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
Traceback (most recent call last):
  File "./ipwndfu", line 47, in <module>
    device = dfu.acquire_device()
  File "/Users/BHNE/Desktop/1033-OTA-Downgrader-master/ipwndfu_public/dfu.py", line 16, in acquire_device
    for device in usb.core.find(find_all=True, idVendor=0x5AC, idProduct=0x1227, backend=backend):
  File "/Users/BHNE/Desktop/1033-OTA-Downgrader-master/ipwndfu_public/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
2019-11-12 10:53:14.738 system_profiler[94264:524575] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode

RowanMcCay, 2019-11-15:
I fixed it by running

`brew unlink libplist`

closed 2019-11-18 by Merculous