MatthewPierson/1033-OTA-Downgrader #8: Failure on restore.sh (error code 11)

NewbieDew, 2019-10-12:
**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Are you futurerestoring an iPhone or an iPad?
iPhone
Please enter device ID (iPhone6,1 or iPhone6,2 only)
iPhone6,1
[Log] Vaild device, continuing
[Log] Getting current APNonce
5184223806800
1bfcdc2860c00debd53e64620f6084b31c678d03
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "5184223806800" to dec:5184223806800 hex:4b70bd18550
[TSSC] manually specified ApNonce to use, parsed "1bfcdc2860c00debd53e64620f6084b31c678d03" to hex:1bfcdc2860c00debd53e64620f6084b31c678d03
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPhone6,1 IS being signed!
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
n
[Log] SHSH saved
[Log] Starting restore process

[Log] No baseband required
[Log] SEP and Baseband copied

[Log] Copying SEP and Baseband
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
INFO: device serial number is DNPLQ0A5FFDP
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/OTA.shsh is done
Found device iPhone6,1 n51ap
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Did set SEP+baseband path and firmware
[WARNING] Failed to read BasebandGoldCertID from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandGoldCertID. This might result in invalid baseband signing status information
[WARNING] Failed to read BasebandSerialNumber from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandSerialNumber size. This might result in invalid baseband signing status information
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] User specified to request only a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Found device in Recovery mode
Device already in Recovery mode
Found device in Recovery mode
Identified device as n51ap, iPhone6,1
Extracting BuildManifest from iPSW
Product version: 10.3.3
Product build: 14G60 Major: 14
Device supports IMG4: true
Got ApNonce from device: 
checking APTicket to be valid for this restore...
Verified ECID in APTicket matches device ECID
checking APTicket to be valid for this restore...
Verified ECID in APTicket matches device ECID
[Error] im4m_buildidentity_check_cb: can't find any identity which matches all hashes inside IM4M
[Error] getBuildIdentityForIM4M: found buildidentity, but can't read information
BuildIdentity selected for restore:
BuildNumber : 14G60
BuildTrain : Greensburg
DeviceClass : n51ap
FDRSupport : NO
RestoreBehavior : Erase
Variant : Customer Erase Install (IPSW)

BuildIdentiy valid for the APTicket:
IM4M isn't valid for any restore with this BuildManifest
This APTicket can't be used for restoring this firmware
IM4M isn't valid for any restore with this BuildManifest
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Extracting filesystem from iPSW
[==================================================] 100.0%
Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (653333 bytes)...
waiting for device to reconnect... 
Getting SepNonce in recovery mode... 1b fc dc 28 60 c0 0d eb d5 3e 64 62 0f 60 84 b3 1c 67 8d 03 
Getting ApNonce in recovery mode... 
[WARNING] Setting bgcolor to green! If you don't see a green screen, then your device didn't boot iBEC correctly
Recovery Mode Environment:
iBoot build-version=iBoot-3406.60.10
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (11640 bytes)...
ramdisk-size=0x10000000
Extracting 058-74940-063.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41583638 bytes)...
Extracting DeviceTree.n51ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (107919 bytes)...
Extracting kernelcache.release.iphone6...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12178427 bytes)...
Trying to fetch new signing tickets
ERROR: Unable to find required ApNonce in parameters
ERROR: Unable to add img4 tags to TSS request
[Error] ERROR: Unable to get signing tickets for SEP

To exit recovery mode, use --exit-recovery

Done: restoring failed.
Failed with errorcode=-11
[Log] Futurerestoring complete

**************** Downgrade complete! Enjoy 10.3.3 =) ****************
**************** Follow me on twitter @mosk_i for help/updates ******
marius@Mariuss-MacBook-Air 1033-OTA-Downgrader-master %

MatthewPierson, 2019-10-12:
Try running install.sh again and see if that fixes the issue. Also what macOS version are you running?

NewbieDew, 2019-10-13:
I've tried multiple times, everything installs correctly (after manually linking some dependencies). I'm on Catalina but will try to make a bootable usb to use Mojave (when I get back home)

MatthewPierson, 2019-10-27:
By the way, the latest version of the script should work on Catalina, its a complete rewrite so maybe you will have better luck now!

NewbieDew, 2019-10-28:
Thank you, it worked perfectly now😁

closed 2019-11-18 by Merculous