MatthewPierson/1033-OTA-Downgrader #9: I'm not very sure if this is the error that is supposed to happen up to 40 times

ilegitimado, 2019-10-12:
[image]

consolelogluke, 2019-10-12:
Doesn't work in a virtual machine. If you don't have a Mac you could use Hackintosh.

0x0luke, 2019-10-12:
If you fully pass the USB controller into a VM, it *SHOULD* work, but I have no idea how that's done in VMware.

consolelogluke, 2019-10-13:
People have tried before with no success

0x0luke, 2019-10-13:
Qwerty got this working with his hackintosh - https://twitter.com/qwertyoruiopz/status/1181875003811418112

consolelogluke, 2019-10-13:
Ooh I interpreted your reply as USB pass through.

ilegitimado, 2019-10-13:
I can connect the iPhone to my Windows or to my VM. I will try to install hackintosh on a spare hard drive I have just for this. Thank you.

closed 2019-10-13 by ilegitimado