MatthewPierson/1033-OTA-Downgrader #14: Script doesn't stop on errors

13XxM1CHA3lxX37, 2019-10-13:
The script doesn't stop on errors like "file not found" and problems with the dependencies (brew link did not work for wget bc of permission problems). 
Had also some problems extracting from my ipsw as my system language was set to Russian and the Terminal did not recognize some Folders because of the cyrillic chracters.

The script just continued as if everything worked.
After I found all errors manually and fixed them everything went fine

consolelogluke, 2019-10-13:
Stopping the script when there’s an error isn’t particularly ideal when using shell scripts from my understanding. You would need to check after every single command what the status of that command was which can get very tedious.

MatthewPierson, 2019-10-13:
I might add a few more checks for specific errors but to add checks for everything is just not worth my time to be honest, when it already outputs when something didn't work (for the most part). As for the issues with cyrillic chracters and some folders, I have no idea how I would address that as I have no experience at all with cyrillic chracters and scripts. If you want to submit a pull request with any changes that you made for cyrillic chracters then feel free!

13XxM1CHA3lxX37, 2019-10-13:
I see your point the output should be enough with some general knowledge. Maybe the issue helps others discover that they need to check the log.
 The cyrillic characters are a very specific issue, fixed it by changing system language.

jeffreyvdhondel, 2019-10-14:
Maybe you could set the LC_CTYPE/LC_ALL to prevent this issue with the cyrillic characters :)

closed 2019-11-18 by Merculous