MatthewPierson/1033-OTA-Downgrader #15: Error starting futurerestore

Nyxest, 2019-10-13:
**************** Downgrade complete! Enjoy 10.3.3 =) ****************
**************** Follow me on twitter @mosk_i for help/updates ******
Nyquiest:1033-OTA-Downgrader-master alex$ ./restore.sh

**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Are you futurerestoring an iPhone or an iPad?
iPhone
Please enter device ID (iPhone6,1 or iPhone6,2 only)
iPhone6,1
[Log] Vaild device, continuing
[Log] Getting current APNonce
4362837131420
060711a8203327cee8c7cabeec7a46e4a4664e9d
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "4362837131420" to dec:4362837131420 hex:3f7cd5b809c
[TSSC] manually specified ApNonce to use, parsed "060711a8203327cee8c7cabeec7a46e4a4664e9d" to hex:060711a8203327cee8c7cabeec7a46e4a4664e9d
[TSSC] opening restore/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPhone6,1 IS being signed!
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
y
Please drag and drop desired folder into this terminal window. (No trailing /)
/Users/Alex/Desktop/iOS\ 10 
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file ... target_directory
[Log] SHSH saved
[Log] Starting restore process

[Log] No baseband required
[Log] SEP and Baseband copied

[Log] Copying SEP and Baseband
cp: 10.3.3.custom/Firmware/Mav7Mav8-7.60.00.Release.bbfw: No such file or directory
cp: 10.3.3.custom/Firmware/all_flash/sep-firmware.n53.RELEASE.im4p: No such file or directory
cp: 10.3.3.custom/Firmware/all_flash/sep-firmware.n51.RELEASE.im4p: No such file or directory
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
dyld: Library not loaded: /rsu/local/opt/openssl/lib/libssl.1.0.0.dylib
  Referenced from: /Users/Alex/Desktop/1033-OTA-Downgrader-master/bin/futurerestore
  Reason: image not found
./restore.sh: line 132: 58821 Abort trap: 6           bin/futurerestore -t shsh/OTA.shsh -s restore/sep-firmware."$device".RELEASE.im4p -m restore/BuildManifest_"$device"_1033_OTA.plist -b restore/Baseband.bbfw -p restore/BuildManifest_"$device"_1033_OTA.plist 10.3.3.custom.ipsw
[Log] Futurerestoring complete

consolelogluke, 2019-10-13:
Are you using macOS Catalina?

Nyxest, 2019-10-13:
Yes but i'm installing a Mojave partition

consolelogluke, 2019-10-13:
Huh, that issue usually happens on macOS Catalina. Did you run `install.sh` and `prep.sh`?

Nyxest, 2019-10-13:
Sure, i did everything like 4 or 5 times but nothing changes

MatthewPierson, 2019-10-13:
Try "brew reinstall openssl" and see if that helps.

closed 2019-11-18 by Merculous