MatthewPierson/1033-OTA-Downgrader #18: rmsigchks.py error

mgfjxxiexiaolong, 2019-10-13:
[image]

MatthewPierson, 2019-10-13:
Try reentering DFU mode and trying pwn.sh again. That is an error that sometimes happens with rmsigchks.py and is usually fixed with a new attempt

13XxM1CHA3lxX37, 2019-10-13:
I get this error everytime I run the rmsignchecks script but it works nevertheless. Just ignore it and continue

mgfjxxiexiaolong, 2019-10-16:
`image not found` when execute './restore.sh'
[image]

mgfjxxiexiaolong, 2019-10-16:
> `image not found` when execute './restore.sh'
> [image]

Fucking successful ! Thanks @MatthewPierson .

closed 2019-10-16 by mgfjxxiexiaolong