MatthewPierson/1033-OTA-Downgrader #22: receiving proxy payload failed

saury, 2019-10-14:
Device: iPad Air 1
MacOS: Mojave

The errors keeps showing after execute `./restore.sh`. Here's info from console:

```
**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Are you futurerestoring an iPhone or an iPad?
iPad
Please enter device ID (iPad4,1 iPad4,2 iPad4,3 iPad4,4 and iPad4,5)
iPad4,1
[Log] Vaild device, continuing
[Log] Getting current APNonce
4899127414352
ed85a1dfc3119de81f661508c3257705f265ecc2
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "4899127414352" to dec:4899127414352 hex:474aabfce50
[TSSC] manually specified ApNonce to use, parsed "ed85a1dfc3119de81f661508c3257705f265ecc2" to hex:ed85a1dfc3119de81f661508c3257705f265ecc2
[TSSC] opening restore/BuildManifest_iPad4,1_1033_OTA.plist
[TSSR] LOG: device iPad4,1 doesn't need a baseband ticket, continuing without requesting a Baseband ticket
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPad4,1 IS being signed!
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
n
[Log] SHSH saved
[Log] Starting restore process

[Log] No baseband required
cp: 10.3.3.custom/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p: No such file or directory
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
INFO: device serial number is DLXLLAH4FK10
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/OTA.shsh is done
Found device iPad4,1 j71ap
[TSSC] opening restore/BuildManifest_iPad4,1_1033_OTA.plist
WARNING: Unable to find BbChipID node
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received

WARNING: user specified not to flash a baseband. This can make the restore fail if the device needs a baseband!
if you added this flag by mistake you can press CTRL-C now to cancel
continuing restore in 10 9 8 7 6 5 4 3 2 1
Found device in Recovery mode
Device already in Recovery mode
Found device in Recovery mode
Identified device as j71ap, iPad4,1
Extracting BuildManifest from iPSW
Product version: 10.3.3
Product build: 14G60 Major: 14
Device supports IMG4: true
Got ApNonce from device: ed 85 a1 df c3 11 9d e8 1f 66 15 08 c3 25 77 05 f2 65 ec c2
checking APTicket to be valid for this restore...
Verified ECID in APTicket matches device ECID
checking APTicket to be valid for this restore...
Verified ECID in APTicket matches device ECID
[Error] im4m_buildidentity_check_cb: can't find any identity which matches all hashes inside IM4M
[Error] getBuildIdentityForIM4M: found buildidentity, but can't read information
BuildIdentity selected for restore:
BuildNumber : 14G60
BuildTrain : Greensburg
DeviceClass : j71ap
FDRSupport : NO
RestoreBehavior : Erase
Variant : Customer Erase Install (IPSW)

BuildIdentiy valid for the APTicket:
IM4M isn't valid for any restore with this BuildManifest
This APTicket can't be used for restoring this firmware
IM4M isn't valid for any restore with this BuildManifest
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Extracting filesystem from iPSW
[==================================================] 100.0%
Extracting iBEC.ipad4.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (673810 bytes)...
waiting for device to reconnect...
Getting SepNonce in recovery mode... 9f 5f 0b 04 02 3c 6e 3b ea cb c8 64 f1 44 c4 85 66 c8 20 38
Getting ApNonce in recovery mode... ed 85 a1 df c3 11 9d e8 1f 66 15 08 c3 25 77 05 f2 65 ec c2
[WARNING] Setting bgcolor to green! If you don't see a green screen, then your device didn't boot iBEC correctly
Recovery Mode Environment:
iBoot build-version=iBoot-3406.60.10
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (22710 bytes)...
ramdisk-size=0x10000000
Extracting 058-74940-063.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41591830 bytes)...
Extracting DeviceTree.j71ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (101147 bytes)...
Extracting kernelcache.release.ipad4...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12259801 bytes)...
Trying to fetch new signing tickets
WARNING: Unable to find BbChipID node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received signing tickets
About to restore device...
Waiting for device...
Device d7a89e7f49ff7caeb74efdd0b58d3346d01babbf is now connected in restore mode...
Connecting now...
Connected to com.apple.mobile.restored, version 14
Device d7a89e7f49ff7caeb74efdd0b58d3346d01babbf has successfully entered restore mode
Hardware Information:
BoardID: 16
ChipID: 35168
UniqueChipID: 4899127414352
ProductionMode: true
Starting FDR listener thread
About to send NORData...
Found firmware path Firmware/all_flash
Getting firmware manifest from build identity
Extracting LLB.ipad4.RELEASE.im4p...
Personalizing IMG4 component LLB...
Extracting applelogo@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@2x~ipad-lightning.s5l8960x.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.j71ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@2x~ipad-lightning.s5l8960x.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.ipad4.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Personalizing IMG4 component RestoreSEP...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Checking filesystems (15)
Checking filesystems (15)
Checking filesystems (15)
ERROR: FDR 0x7fccdf765900 receiving proxy payload failed: Operation now in progress
ERROR: Unable to receive message from FDR 0x7fccdf765900 (-2). 0/2 bytes
ERROR: FDR 0x7fcce2000170 receiving proxy payload failed: Operation now in progress
ERROR: Unable to receive message from FDR 0x7fcce2000170 (-2). 0/2 bytes
ERROR: FDR 0x7fccdf765900 receiving proxy payload failed: Operation now in progress
ERROR: Unable to receive message from FDR 0x7fccdf765900 (-2). 0/2 bytes
ERROR: FDR 0x7fcce1e7cd70 receiving proxy payload failed: Operation now in progress
ERROR: Unable to receive message from FDR 0x7fcce1e7cd70 (-2). 0/2 bytes
ERROR: FDR 0x7fccdf765900 receiving proxy payload failed: Operation now in progress
```

MatthewPierson, 2019-10-15:
I think is is a futurerestore error more then an error with my stuff from a quick google search. Try googling that error to find a fix

saury, 2019-10-16:
Fixed it by reseting my device. 
Anyone who shall meet this kind issue please reset your device to make sure it's clean and not sign in with your apple id.

closed 2019-10-16 by saury