MatthewPierson/1033-OTA-Downgrader #23: Failed with errorcode=-11 Unable to open device in restore mode

vaczi001, 2019-10-14:
When executing restore.sh it's always stuck at:
```
About to restore device...
Waiting for device...
ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
[Error] ERROR: Unable to restore device
Done: restoring failed.
Failed with errorcode=-11
```
Phone is in Restore mode and iTunes recognizes it.
Looks like the usbmuxd is broken.
Tried on Catalina, Mojave, a 2nd computer too, same error message.

MatthewPierson, 2019-10-14:
If iTunes recognizes the device then it is not in restore mode, its in recovery or (soft) DFU mode. Try entering pwndfu mode and restoring again

vaczi001, 2019-10-15:
Okay I got it working now, the problem was every time after pwn.sh the phone won't go to pwndfu mode, just to recovery. After just a restart everything went alright. Thanks a lot, finally on iOS 10.😊

And I did it on Catalina so I can confirm it's working. Just need the system drive to be r-w when executing install & prep.sh

closed 2019-10-15 by vaczi001