MatthewPierson/1033-OTA-Downgrader #28: High Sierra supported or not anybody tested?

omeryarar, 2019-10-16:
let me know plz if hs is supported

MatthewPierson, 2019-10-16:
As the readme states only Mojave is fully supported. Catalina has partial support, with a workaround but no other os.

consolelogluke, 2019-10-17:
High Sierra fails at the last stage (install.sh). Some guy got it working in #10 but @MatthewPierson removed the download because he didn't think it fixed anything

MatthewPierson, 2019-10-17:
I removed it because it had no actual changes. He never replied with what he "changed" to make it work so as far as I'm concerned there was never HS support.

closed 2019-10-17 by MatthewPierson

Lengran1992, 2019-10-17:
Same issue on High Sierra

consolelogluke, 2019-10-17:
@MatthewPierson I’m not going to start arguing about it but I do want to mention that it *did* work though

MatthewPierson, 2019-10-17:
Then as I said before, show me what was changed. I'm fine with admitting I was wrong and apologizing if I'm shown what the changes were that fixed it on HS.

consolelogluke, 2019-10-17:
Do you have the modified version? I deleted it after using it.

Lengran1992, 2019-10-18:
Solved!
No modified version,just replace with old version futurestore.