MatthewPierson/1033-OTA-Downgrader #29: Error on High Sierra

Lengran1992, 2019-10-17:
dyld: Library not loaded: /rsu/local/opt/usbmuxd/lib/libusbmuxd.4.dylib
Referenced from: /Users/lengran/Downloads/1033-OTA-Downgrader-master/bin/futurerestore
Reason: Incompatible library version: futurerestore requires version 6.0.0 or later, but libusbmuxd.4.dylib provides version 5.0.0
./restore.sh: line 171: 83944 Abort trap: 6           bin/futurerestore --exit-recovery

consolelogluke, 2019-10-17:
Duplicate of #10

omeryarar, 2019-10-17:
matty hasnt a high sierra system he has mojave if somebody pull a request for high sierra he will check and then maybe he releases if it is safer

Lengran1992, 2019-10-18:
Solved!
Replace old version futurestore~

omeryarar, 2019-10-18:
can u give more details and did downgrade succeed?

Lengran1992, 2019-10-18:
@langyanduan 

> can u give more details and did downgrade succeed?

https://www.dropbox.com/s/2tcdj40d1q9o2xu/futurerestore?dl=0
Download&Replace
Path: .../1033-OTA-Downgrader-master/bin

MatthewPierson, 2019-10-18:
That futurerestore binary will not work for those who don't have all needed the dependencies installed, which is part of why the original error occured. It may work for you so go ahead and try but just know that it isnt a 100% fix for everyone.

Lengran1992, 2019-10-18:
ok~
I'll cancel download-link later.
Anyway,grateful for you excellent tool.

mittovipi, 2019-10-19:
Error here
Device supports IMG4: true
Get ApNonce from the device:
It does not read ApNonce from the device

jejer, 2019-10-19:
can not get apnonce during restore using above old version futurerestore.
```
Extracting iBEC.ipad4.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (673810 bytes)...
waiting for device to reconnect...
Getting SepNonce in recovery mode... c5 1f d1 fc 57 68 9c 50 9b 3e e1 7b e3 3c 6d a3 32 5a b5 9d
Getting ApNonce in recovery mode...
[WARNING] Setting bgcolor to green! If you don't see a green screen, then your device didn't boot iBEC correctly
Recovery Mode Environment:
iBoot build-version=iBoot-3406.60.10
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (22710 bytes)...
ramdisk-size=0x10000000
Extracting 058-74940-063.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41591830 bytes)...
Extracting DeviceTree.j71ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (101147 bytes)...
Extracting kernelcache.release.ipad4...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12259801 bytes)...
Trying to fetch new signing tickets
WARNING: Unable to find BbChipID node
ERROR: Unable to find required ApNonce in parameters
ERROR: Unable to add img4 tags to TSS request
[Error] ERROR: Unable to get signing tickets for SEP

To exit recovery mode, use --exit-recovery

Done: restoring failed.
Failed with errorcode=-11
[Log] Futurerestore failed
[Log] Exiting recovery mode
```

MatthewPierson, 2019-10-19:
@jejer I'm not going to help you fix an issue with something someone else posted for you to try. As I said, "It may work for you so go ahead and try but just know that it isnt a 100% fix for everyone.".

jejer, 2019-10-19:
@MatthewPierson I know, just document the failure here to save other's time if they want to try.
It's really time-consuming to enter pwned dfu mode on my iPad air.
Maybe some other talent will help to fix it.
Thank you and @Lengran1992

jejer, 2019-10-19:
finally, get futurerestore works on high sierra.
Just compile and use https://github.com/merculous/futurerestore
I got this idea from https://github.com/Merculous/1033-OTA-Downgrader/
Downgrade complete!

omeryarar, 2019-10-20:
if u guide i will try it

luzeyang, 2019-10-21:
> finally, get futurerestore works on high sierra.
> Just compile and use https://github.com/merculous/futurerestore
> run `ln -s /usr/local/Cellar/openssl/1.0.2t/lib/pkgconfig/* /usr/local/lib/pkgconfig/` to fix compile issue.
> I got this idea from https://github.com/Merculous/1033-OTA-Downgrader/
> Downgrade complete!

Can you elaborate on the steps, thank you

omeryarar, 2019-10-21:
the question is what Matty thinks about this if he says yes im gonna try

mittovipi, 2019-10-24:
https://github.com/MatthewPierson/1033-OTA-Downgrader/issues/29#issuecomment-543167399

I fixed a bug on high sierra.  100% working perfectly

yohnz, 2019-10-25:
> [#29 (comment)](https://github.com/MatthewPierson/1033-OTA-Downgrader/issues/29#issuecomment-543167399)
> 
> I fixed a bug on high sierra. 100% working perfectly

how to fix?

mittovipi, 2019-10-28:
DOWNGRADE IOS 10.3.3 CHIP A7 FINAL

LINK SIERRA: bit.ly/videosierra

LINK HIGH SIERRA: bit.ly/videofinalhighsierradowngrade

MatthewPierson, 2019-10-28:
@mittovipi how many fucking times do I need to say this, post the files not bit.ly links. I'm going to delete your comments because you can't seem to understand that. I've asked you so many times to explain how you did it or post the changes but you keep posting those links. Surprisingly again all I can find are fucking cosmetic changes, not functional changes so unless you can explain what you think you did to add High Sierra support I'm going to keep deleting your comments

mittovipi, 2019-10-28:
problem in rsu file.
This is just a personalization to make it easier for me to understand. Sorry for editing it

omeryarar, 2019-10-28:
i can use checkm8 exploit on Hs and i can use futurerestore on hs but cant use this script and that is the worst thing

closed 2019-11-18 by Merculous