MatthewPierson/1033-OTA-Downgrader #30: Error on pwn.sh and restore.sh

luquillas02, 2019-10-17:
Hey so all of the steps work until pwn.sh, which seems to work, but some python dylib crashes in the background. After reopening it and trying again it seems to work. Now on the restore.sh this is what I get: 

**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Are you futurerestoring an iPhone or an iPad?
iPhone
Please enter device ID (iPhone6,1 or iPhone6,2 only)
iPhone6,2
[Log] Vaild device, continuing
[Log] Getting current APNonce
4357492002972
6b81a2c3cdf87404dee28330f7fcb0ee62c425a1
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "4357492002972" to dec:4357492002972 hex:3f68ec3509c
[TSSC] manually specified ApNonce to use, parsed "6b81a2c3cdf87404dee28330f7fcb0ee62c425a1" to hex:6b81a2c3cdf87404dee28330f7fcb0ee62c425a1
[TSSC] opening restore/BuildManifest_iPhone6,2_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPhone6,2 IS being signed!
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
n
[Log] SHSH saved
[Log] Starting restore process

[Log] No baseband required
[Log] SEP and Baseband copied

[Log] Copying SEP and Baseband
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
dyld: Library not loaded: /rsu/local/opt/openssl/lib/libssl.1.0.0.dylib
  Referenced from: /Users/lucas/Downloads/1033-OTA-Downgrader-master/bin/futurerestore
  Reason: image not found
./restore.sh: line 132: 59799 Abort trap: 6           bin/futurerestore -t shsh/OTA.shsh -s restore/sep-firmware."$device".RELEASE.im4p -m restore/BuildManifest_"$device"_1033_OTA.plist -b restore/Baseband.bbfw -p restore/BuildManifest_"$device"_1033_OTA.plist 10.3.3.custom.ipsw
[Log] Futurerestoring complete

**************** Downgrade complete! Enjoy 10.3.3 =) ****************
**************** Follow me on twitter @mosk_i for help/updates ******
lucas@Lucass-MacBook-Air 1033-OTA-Downgrader-master %

MatthewPierson, 2019-10-17:
What macOS version? If its Catalina then there is a workaround you can use

luquillas02, 2019-10-18:
Yes I am on Catalina. What’s the workaround? :)

MatthewPierson, 2019-10-18:
The steps are in #26

closed 2019-11-18 by Merculous