MatthewPierson/1033-OTA-Downgrader #31: Faild

azzaba, 2019-10-19:
Hi thx for your work bro :) i have ipad air and fresh install of  moajve ,i try to restore manu times but i get always the same error 
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (22710 bytes)...
ramdisk-size=0x10000000
Extracting 058-74940-063.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41591830 bytes)...
Extracting DeviceTree.j71ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (101147 bytes)...
Extracting kernelcache.release.ipad4...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12259801 bytes)...
Trying to fetch new signing tickets
WARNING: Unable to find BbChipID node
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received signing tickets
About to restore device... 
Waiting for device...
Device 62457a010ab91db71acc606c9c83b30919e44d5f is now connected in restore mode...
Connecting now...
ERROR: Unable to open device in restore mode
[Error] ERROR: Unable to restore device
Done: restoring failed.
Failed with errorcode=-11
[Log] Futurerestore failed
[Log] Exiting recovery mode
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
Exiting recovery mode
ERROR: Device is in an invalid state
[INFO] 32-bit device detected
futurerestore init done
[Error] can't set auto-boot, when device isn't in recovery mode
libc++abi.dylib: terminating with uncaught exception of type int
./restore.sh: line 171:  2141 Abort trap: 6           bin/futurerestore --exit-recovery

[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
MacBook-Air-de-MAC:1033 mac$

Edythator, 2019-10-19:
Having the same issue here, running on an iPad4,1.

consolelogluke, 2019-10-19:
Exit DFU mode (hold power button + home button for 10 seconds) then enter DFU mode again. Run `pwn.sh` until it succeeds and try running `restore.sh` again.

closed 2019-10-21 by azzaba