MatthewPierson/1033-OTA-Downgrader #37: ./restore.sh: line 151: bin/futurerestore: No such file or directory

yohnz, 2019-10-23:
Mojave 10.14 (18A391)
```
 ./restore.sh                                                                                                                                                             !10871
**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Are you futurerestoring an iPhone or an iPad?
iPad
Please enter device ID (iPad4,1 iPad4,2 iPad4,3 iPad4,4 and iPad4,5)
iPad4,4
[Log] Vaild device, continuing
[Log] Getting current APNonce

./restore.sh: line 62: bin/tsschecker: No such file or directory
mv: rename shsh/* to shsh/OTA.shsh: No such file or directory
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
n
[Log] SHSH saved
[Log] Starting restore process

[Log] No baseband required
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
./restore.sh: line 151: bin/futurerestore: No such file or directory
[Log] Futurerestore failed
[Log] Exiting recovery mode
./restore.sh: line 164: bin/futurerestore: No such file or directory

[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
```

MatthewPierson, 2019-10-23:
Did you run install.sh and prep.sh before restore.sh?

yohnz, 2019-10-24:
> Did you run install.sh and prep.sh before restore.sh?

yes, I tried several times, follow the readme.

geekchen404, 2019-10-25:
run 'brew upgrade openssl'

geekchen404, 2019-10-25:
> > 您是否在restore.sh之前运行过install.sh和prep.sh？
> 
> 是的，我尝试了几次，遵循自述文件。

run 'brew upgrade openssl'

MatthewPierson, 2019-10-27:
Please try the lastest version of the script. It rewrites everything and should fix this issue

closed 2019-11-18 by Merculous