MatthewPierson/1033-OTA-Downgrader #39: futurerestore requires version 6.0.0 or later, but libusbmuxd.4.dylib provides version 5.0.0

Netfliics, 2019-10-25:
i keep getting this:

**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Are you futurerestoring an iPhone or an iPad?
iPad
Please enter device ID (iPad4,1 iPad4,2 iPad4,3 iPad4,4 and iPad4,5)
iPad4,4
[Log] Vaild device, continuing
[Log] Getting current APNonce
3359957646684
efd3060f847c13dbd79b955cc9761a76bb6b9653
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "3359957646684" to dec:3359957646684 hex:30e4d14fd5c
[TSSC] manually specified ApNonce to use, parsed "efd3060f847c13dbd79b955cc9761a76bb6b9653" to hex:efd3060f847c13dbd79b955cc9761a76bb6b9653
[TSSC] opening restore/BuildManifest_iPad4,4_1033_OTA.plist
[TSSR] LOG: device iPad4,4 doesn't need a baseband ticket, continuing without requesting a Baseband ticket
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPad4,4 IS being signed!
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
n
[Log] SHSH saved
[Log] Starting restore process

[Log] No baseband required
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
dyld: Library not loaded: /rsu/local/opt/usbmuxd/lib/libusbmuxd.4.dylib
  Referenced from: /Users/riccardocardarelli/Desktop/1033-OTA-Downgrader-master/bin/futurerestore
  Reason: Incompatible library version: futurerestore requires version 6.0.0 or later, but libusbmuxd.4.dylib provides version 5.0.0
./restore.sh: line 153: 23214 Abort trap: 6           bin/futurerestore -t shsh/OTA.shsh -s restore/sep-firmware."$device".RELEASE.im4p -m restore/BuildManifest_"$device"_1033_OTA.plist --no-baseband 10.3.3.custom.ipsw
[Log] Futurerestore failed
[Log] Exiting recovery mode
dyld: Library not loaded: /rsu/local/opt/usbmuxd/lib/libusbmuxd.4.dylib
  Referenced from: /Users/riccardocardarelli/Desktop/1033-OTA-Downgrader-master/bin/futurerestore
  Reason: Incompatible library version: futurerestore requires version 6.0.0 or later, but libusbmuxd.4.dylib provides version 5.0.0
./restore.sh: line 171: 23215 Abort trap: 6           bin/futurerestore --exit-recovery

[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED

MatthewPierson, 2019-10-25:
What macOS version are you running?

Netfliics, 2019-10-25:
High Sierra, right now i’m downloading Mojave

MatthewPierson, 2019-10-25:
This is a known issue that I'm currently trying to fix. Let me know if Mojave works for you

consolelogluke, 2019-10-25:
It can be fixed by patching an older version of futurerestore

Netfliics, 2019-10-26:
Okay, thank you for replying, i’m installing it right now

Netfliics, 2019-10-26:
IT WORKED

MatthewPierson, 2019-10-26:
What did you do to fix it?

omeryarar, 2019-10-26:
I think Mojave worked for him.

letinmore, 2019-10-26:
What can I do to solve this error? Sadly my old iMac can't go beyond High Sierra, and I'm stuck with this same error. Do I need an older version of futurerestore (not sure if it's possible to find one) or can I use Linux as an alternative?

consolelogluke, 2019-10-26:
You can use Mojave Patcher or wait for an update.

You can also use an older version of futurerestore, I think you can just look at the release history for futurerestore and you’ll see a version that has a change saying that it updated libusbmux. DM me on Twitter (@ConsoleLogLuke) if you want me to help you get it working.

closed 2019-11-18 by Merculous