MatthewPierson/1033-OTA-Downgrader #41: ./restore.sh  WARNING: Unable to find BbChipID node

icode-cn, 2019-10-26:
**************** Matty's iPhone 5s 10.3.3 OTA Downgrader ****************
[Log] Removing old files
Are you futurerestoring an iPhone or an iPad?
iPad
Please enter device ID (iPad4,1 iPad4,2 iPad4,3 iPad4,4 and iPad4,5)
iPad4,1
[Log] Vaild device, continuing
[Log] Getting current APNonce
5176113643056
3a88b7c3802f2f0510abc432104a15ebd8bd7154
Version: c0b554e83f54d39d90cac9791160bf2ccb062aed - 355
[TSSC] manually specified ECID to use, parsed "5176113643056" to dec:5176113643056 hex:4b5286a3e30
[TSSC] manually specified ApNonce to use, parsed "3a88b7c3802f2f0510abc432104a15ebd8bd7154" to hex:3a88b7c3802f2f0510abc432104a15ebd8bd7154
[TSSC] opening restore/BuildManifest_iPad4,1_1033_OTA.plist
[TSSR] LOG: device iPad4,1 doesn't need a baseband ticket, continuing without requesting a Baseband ticket
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

Build 14G60 for device iPad4,1 IS being signed!
Do you want to save a copy of the OTA SHSH to somewhere on your computer? (y/n)
n
[Log] SHSH saved
[Log] Starting restore process

[Log] No baseband required
cp: 10.3.3.custom/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p: No such file or directory
[Log] SEP and Baseband copied

[Log] Cleaning up un-needed files
[Log] Clean up done
[Log] Starting futurerestore
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
INFO: device serial number is DMRMT3CMFK14
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/OTA.shsh is done
Found device iPad4,1 j71ap
[TSSC] opening restore/BuildManifest_iPad4,1_1033_OTA.plist
WARNING: Unable to find BbChipID node
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received

WARNING: user specified not to flash a baseband. This can make the restore fail if the device needs a baseband!
if you added this flag by mistake you can press CTRL-C now to cancel
continuing restore in 10 9 8 7 6 5 4 3 2 1 
Found device in Recovery mode
Device already in Recovery mode
Found device in Recovery mode
Identified device as j71ap, iPad4,1
Extracting BuildManifest from iPSW
Product version: 10.3.3
Product build: 14G60 Major: 14
Device supports IMG4: true
Got ApNonce from device: 3a 88 b7 c3 80 2f 2f 05 10 ab c4 32 10 4a 15 eb d8 bd 71 54 
checking APTicket to be valid for this restore...
Verified ECID in APTicket matches device ECID
checking APTicket to be valid for this restore...
Verified ECID in APTicket matches device ECID
[Error] im4m_buildidentity_check_cb: can't find any identity which matches all hashes inside IM4M
[Error] getBuildIdentityForIM4M: found buildidentity, but can't read information
BuildIdentity selected for restore:
BuildNumber : 14G60
BuildTrain : Greensburg
DeviceClass : j71ap
FDRSupport : NO
RestoreBehavior : Erase
Variant : Customer Erase Install (IPSW)

BuildIdentiy valid for the APTicket:
IM4M isn't valid for any restore with this BuildManifest
This APTicket can't be used for restoring this firmware
IM4M isn't valid for any restore with this BuildManifest
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Extracting filesystem from iPSW
[==================================================] 100.0%
Extracting iBEC.ipad4.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (673810 bytes)...
waiting for device to reconnect... 
Getting SepNonce in recovery mode... df 6f 69 bd 27 24 98 61 84 ce 70 78 24 43 b7 67 d3 27 bb 47 
Getting ApNonce in recovery mode... 3a 88 b7 c3 80 2f 2f 05 10 ab c4 32 10 4a 15 eb d8 bd 71 54 
[WARNING] Setting bgcolor to green! If you don't see a green screen, then your device didn't boot iBEC correctly
Recovery Mode Environment:
iBoot build-version=iBoot-4513.270.14
iBoot build-style=RELEASE
Sending RestoreLogo...
Extracting applelogo@2x~ipad.s5l8960x.im4p...
Personalizing IMG4 component RestoreLogo...
Sending RestoreLogo (22710 bytes)...
ramdisk-size=0x10000000
Extracting 058-74940-063.dmg...
Personalizing IMG4 component RestoreRamDisk...
Sending RestoreRamDisk (41591830 bytes)...
Extracting DeviceTree.j71ap.im4p...
Personalizing IMG4 component RestoreDeviceTree...
Sending RestoreDeviceTree (101147 bytes)...
Extracting kernelcache.release.ipad4...
Personalizing IMG4 component RestoreKernelCache...
Sending RestoreKernelCache (12259801 bytes)...
Trying to fetch new signing tickets
**WARNING: Unable to find BbChipID node**
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Received signing tickets
About to restore device... 
Waiting for device...

ERROR: Unable to connect to device in restore mode
ERROR: Unable to open device in restore mode
[Error] ERROR: Unable to restore device
Done: restoring failed.
Failed with errorcode=-11
[Log] Futurerestore failed
[Log] Exiting recovery mode
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
Exiting recovery mode
INFO: device serial number is DMRMT3CMFK14
[INFO] 64-bit device detected
futurerestore init done
Done

[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED
[ERROR] FUTURERESTORE FAILED

How can I fix it?

MatthewPierson, 2019-10-27:
Please try again with latest version of the script. Its been rewritten so hopefully it should fix your issue

chenyinfu, 2019-10-28:
Good.Try again.....

icode-cn, 2019-10-28:
> Please try again with latest version of the script. Its been rewritten so hopefully it should fix your issue

It works, thank you very much!

icode-cn, 2019-10-28:
> Good.Try again.....

OK.

closed 2019-11-18 by Merculous