MatthewPierson/1033-OTA-Downgrader #44: Fail code=-3

musab7777, 2019-10-28:
Hi
Thank you for hard work and great scerpt.
I have this issue. I tried the new scerpt everything was great except in the end it shows me this error:
Fail code=-3
Failed with errorcode=-3
I have ipad mini 2
High Sierra.

Thank you.

linkway321, 2019-10-29:
I also encountered the same problem.
iPad4,4 mini2
macos mojave 10.14.6

musab7777, 2019-10-29:
Same your device ipad mini 2 wifi 
ipad4,4

zwb1990, 2019-10-29:
iPad4,4 mini2   Fail code=-3

MatthewPierson, 2019-10-29:
Please show me the full error. I can't do anything with just "Fail code= -3"

musab7777, 2019-10-29:
WARNING: user specified not to flash a baseband. This can make the restore fail if the device needs a baseband!
if you added this flag by mistake you can press CTRL-C now to cancel
continuing restore in 10 9 8 7 6 5 4 3 2 1 
Found device in DFU mode
[Error] unsupported device mode, please put device in recovery mode or normal mode
[Error] Fail code=-3
Failed with errorcode=-3
Cleaning up :D
dummy_file
iBEC.im4p
iBEC.ipad4b.RELEASE.im4p
iBEC.prepatched
iBEC.raw
Mav7Mav8-7.60.00.Release.bbfw
sep-firmware.j85.RELEASE.im4p
ipsw/._058-75094-062.dmg
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu
ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p
ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p
ipsw/Firmware/all_flash/glyphplugin@2x~ipad-lightning.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j86ap.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j86ap.im4p
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j72.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j71ap.im4p
ipsw/Firmware/all_flash/batterylow1@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j72.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j73ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j73.RELEASE.im4p
ipsw/Firmware/all_flash/iBoot.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/batterylow1@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j73ap.im4p
ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j71ap.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j73.RELEASE.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p
ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/glyphplugin@2x~ipad-lightning.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/058-75094-062.dmg
ipsw/kernelcache.release.ipad4
ipsw/kernelcache.release.ipad4b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/BuildManifest.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

MatthewPierson, 2019-10-29:
I know what the issue is. Give me an hour or so and I will push a fix for it that you can try out. Will reply to this comment and @ you when its out @musab7777

musab7777, 2019-10-29:
thank you sir really I appreciate your help and I'm waiting for you.
Regards,

MatthewPierson, 2019-10-29:
@musab7777 @zwb1990 @linkway321 Can you all please try again with the latest version? I should have fixed the issue with this commit but please let me know if it works or not.

musab7777, 2019-10-29:
Thank you I will try it now

musab7777, 2019-10-29:
I tried still same issue.
And also I want to let you know 
When the script start here 1 2 3 4 ....... for running futurerestore
My iPad still on black screenI mean DFU mode didn’t go to recovery mode

MatthewPierson, 2019-10-29:
Can you show a full log of the entire script?

MatthewPierson, 2019-10-29:
Also the screen will stay black, with just the backlight on, until the screen turns green and the restore begins.

musab7777, 2019-10-29:
cd /Users/hello/Desktop/1033-OTA-Downgrader-master 
033-OTA-Downgrader-master ./restore.sh iPad_64bit_10.3.3_14G60_Restore.ipsw
[+] Installing dependencies
Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libplist 2.0.0_1 is already installed and up-to-date
To reinstall 2.0.0_1, run `brew reinstall libplist`
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID: CPRV: CPFM:SCEP: BDID: ECID: IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Patches have already been applied. Exiting.
We seem to be in pwned DFU mode!
[+] Build folder exists! Updating only!
From https://github.com/merculous/futurerestore
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/tihmstar/img4tool
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/tihmstar/libfragmentzip
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/libimobiledevice/libimobiledevice
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/libimobiledevice/libirecovery
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/tihmstar/liboffsetfinder64
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/libimobiledevice/libusbmuxd
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/s0uthwest/tsschecker
 * branch            master     -> FETCH_HEAD
Already up to date.
shsh/apnonce.shsh2
shsh
downgrade.ipsw
Killing iTunes as this will be quite annoying with what we are going to do.
No matching processes belonging to you were found
Archive:  iPad_64bit_10.3.3_14G60_Restore.ipsw
  inflating: ipsw/058-74940-063.dmg  
  inflating: ipsw/058-75094-062.dmg  
  inflating: ipsw/._058-75094-062.dmg  
  inflating: ipsw/058-75381-062.dmg  
  inflating: ipsw/BuildManifest.plist  
   creating: ipsw/Firmware/
   creating: ipsw/Firmware/all_flash/
  inflating: ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p  
[... 282 lines cut ...]
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/058-75094-062.dmg
ipsw/kernelcache.release.ipad4
ipsw/kernelcache.release.ipad4b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/BuildManifest.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

MatthewPierson, 2019-10-29:
Can you try again with the latest update but this time re-enter DFU mode so reboot the device and enter DFU again. I think the fact you are already in pwndfu when the script starts might be causing issues

musab7777, 2019-10-29:
Ok I will try it now 🌹

musab7777, 2019-10-29:
Sorry sir but still same error ❤️

MatthewPierson, 2019-10-29:
Try deleting everything and starting fresh. Something might have been corrupted at some point, try downloading a new ipsw as well. If that doesn't fix it then I've got no clue whats wrong and will have to look into it more

musab7777, 2019-10-29:
I always delete everything and start fresh because you try to fix it and you upload new files so I deleted everything and also the ipsw I deleted and I download it again then I tried to use the script with the latest update but still I got same error 😫.

Thank you for your hard work sir🌹

linkway321, 2019-10-29:
ipad air 4,2 
[==================================================] 100.0%
[========================================          ] 78.4%Version: 3135d5ce73f7cf7229c373db2f5c333a1c15196b - 359
[TSSC] manually specified ECID to use, parsed "4911456853840" to dec:4911456853840 hex:47789a41f50
[TSSC] manually specified ApNonce to use, parsed "5d39a9b70368e08f50a4df8fae6a1a431cf91c6b" to hex:5d39a9b70368e08f50a4df8fae6a1a431cf91c6b
[TSSC] opening manifests/BuildManifest_iPad4,2_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

iOS 10.3.3 for device iPad4,2 IS being signed!
4911456853840_iPad4,2_10.3.3-14G60_5d39a9b70368e08f50a4df8fae6a1a431cf91c6b.shsh -> shsh/apnonce.shsh2
Done prepping files! Time to downgrade!!!
****RESTORING!****
Waiting for device to reconnect...
Version: 7e66824796f697ea54597dadbdba462382f004b9 - 247
Odysseus support: no
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/apnonce.shsh2 is done
Found device iPad4,2 j72ap
[TSSC] opening manifests/BuildManifest_iPad4,2_1033_OTA.plist
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Did set SEP+baseband path and firmware
[WARNING] Failed to read BasebandGoldCertID from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandGoldCertID. This might result in invalid baseband signing status information
[WARNING] Failed to read BasebandSerialNumber from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandSerialNumber size. This might result in invalid baseband signing status information
[TSSC] opening manifests/BuildManifest_iPad4,2_1033_OTA.plist
[TSSR] User specified to request only a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Found device in DFU mode
[Error] unsupported device mode, please put device in recovery mode or normal mode
[Error] Fail code=-3
Failed with errorcode=-3
Cleaning up :D
dummy_file
iBEC.im4p
iBEC.img4
iBEC.ipad4.RELEASE.im4p
iBEC.prepatched
iBEC.raw
Mav7Mav8-7.60.00.Release.bbfw
sep-firmware.j72.RELEASE.im4p
ipsw/._058-75094-062.dmg
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu
ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p
ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p
[... 35 lines cut ...]
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/058-75094-062.dmg
ipsw/kernelcache.release.ipad4
ipsw/kernelcache.release.ipad4b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/BuildManifest.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
yanningdeMacBook-Air:1033-OTA-Downgrader-master yanning$

MatthewPierson, 2019-10-29:
@linkway321 Im testing a private fix now, will let you know if it works or not.

MatthewPierson, 2019-10-29:
Error code -3 should be fixed now @linkway321 @musab7777

linkway321, 2019-10-29:
I try to download again, thank you.

linkway321, 2019-10-29:
perfect，thank you

kaiseryeahhh, 2019-10-29:
Same issue here, first this morning
Found device in DFU mode
[Error] unsupported device mode, please put device in recovery mode or normal mode
[Error] Fail code=-3
Failed with errorcode=-3
Cleaning up

With the new commit

Done prepping files! Time to downgrade!!!
****RESTORING!****
Waiting for device to reconnect...
Version: 7e66824796f697ea54597dadbdba462382f004b9 - 247
Odysseus support: no
ERROR: Device is in an invalid state
[INFO] 32-bit device detected
[Error] can't init, no device found
Failed with errorcode=-3
Cleaning up :D

Catalina

linkway321, 2019-10-29:
[Error] ERROR: Unable to restore device
Done: restoring failed.
Failed with errorcode=-11
Cleaning up :D
dummy_file
iBSS.im4p
iBSS.img4
iBSS.ipad4b.RELEASE.im4p
iBSS.prepatched
iBSS.raw
iBEC.im4p
iBEC.img4
iBEC.ipad4b.RELEASE.im4p
iBEC.prepatched
iBEC.raw
Mav7Mav8-7.60.00.Release.bbfw
sep-firmware.j85.RELEASE.im4p
downgrade/058-75094-062.dmg
downgrade
ipsw/._058-75094-062.dmg
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu
ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p
ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p
ipsw/Firmware/all_flash/glyphplugin@2x~ipad-lightning.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j86ap.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j86ap.im4p
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j72.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j71ap.im4p
ipsw/Firmware/all_flash/batterylow1@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j72.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j73ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j73.RELEASE.im4p
ipsw/Firmware/all_flash/iBoot.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/batterylow1@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j73ap.im4p
ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j71ap.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j73.RELEASE.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p
ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/glyphplugin@2x~ipad-lightning.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/058-75094-062.dmg
ipsw/kernelcache.release.ipad4
ipsw/kernelcache.release.ipad4b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/BuildManifest.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

lx32056127, 2019-10-29:
> Same issue here, first this morning
> Found device in DFU mode
> [Error] unsupported device mode, please put device in recovery mode or normal mode
> [Error] Fail code=-3
> Failed with errorcode=-3
> Cleaning [...]

I have the same problem.
iPhone6,1
Mojave
Run with root restore.sh 

I tried many times, but each time I failed.

Suspicious 
`
  adding: kernelcache.release.iphone8b (deflated 6%)
[==================================================] 100.0%
ERROR: Unable to connect to device
Version: 3135d5ce73f7cf7229c373db2f5c333a1c15196b - 359
`

the following is log 
=================================================================
img4tool version: 0.152-479a551ea5127e013086aca5e3a6af061f8e2a8a
Created IM4P file at iBSS.im4p
img4tool version: 0.152-479a551ea5127e013086aca5e3a6af061f8e2a8a
Created IM4P file at iBEC.im4p
Version: 3135d5ce73f7cf7229c373db2f5c333a1c15196b - 359
[TSSC] manually specified ECID to use, parsed "4629039364784" to dec:4629039364784 hex:435c83f3ab0
[TSSC] opening manifests/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

iOS 10.3.3 for device iPhone6,1 IS being signed!
shsh/4629039364784_iPhone6,1_10.3.3-14G60_75a4dff692b5fec977b7bb4d7a04896a1f5e1e0e.shsh2 -> shsh/stitch.shsh2
img4tool version: 0.152-479a551ea5127e013086aca5e3a6af061f8e2a8a
Created IMG4 file at iBSS.img4
img4tool version: 0.152-479a551ea5127e013086aca5e3a6af061f8e2a8a
Created IMG4 file at iBEC.img4
iBSS.img4 -> ipsw/Firmware/dfu/iBSS.iphone6.RELEASE.im4p
iBEC.img4 -> ipsw/Firmware/dfu/iBEC.iphone6.RELEASE.im4p
  adding: 058-74917-062.dmg (deflated 1%)
  adding: 058-74940-063.dmg (deflated 17%)
  adding: 058-75381-062.dmg (deflated 17%)
  adding: BuildManifest.plist (deflated 95%)
  adding: Firmware/ (stored 0%)
  adding: Firmware/aopfw.im4p (deflated 45%)
  adding: Firmware/usr/ (stored 0%)
  adding: Firmware/usr/local/ (stored 0%)
  adding: Firmware/usr/local/standalone/ (stored 0%)
  adding: Firmware/dfu/ (stored 0%)
  adding: Firmware/dfu/iBEC.n69.RELEASE.im4p.plist (deflated 26%)
  adding: Firmware/dfu/iBSS.n69u.RELEASE.im4p.plist (deflated 26%)
  adding: Firmware/dfu/iBSS.iphone6.RELEASE.im4p (deflated 51%)
  adding: Firmware/dfu/iBEC.n69.RELEASE.im4p (deflated 0%)
  adding: Firmware/dfu/iBEC.n69u.RELEASE.im4p.plist (deflated 26%)
  adding: Firmware/dfu/iBEC.n69u.RELEASE.im4p (deflated 0%)
  adding: Firmware/dfu/iBSS.iphone6.RELEASE.im4p.plist (deflated 26%)
  adding: Firmware/dfu/iBEC.iphone6.RELEASE.im4p (deflated 49%)
  adding: Firmware/dfu/iBSS.n69.RELEASE.im4p.plist (deflated 26%)
  adding: Firmware/dfu/iBEC.iphone6.RELEASE.im4p.plist (deflated 28%)
[... 214 lines cut ...]
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s8003.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.n53.RELEASE.im4p
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/LLB.n69u.RELEASE.im4p
ipsw/Firmware/all_flash
ipsw/Firmware/Mav10-5.62.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav10-5.62.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/kernelcache.release.iphone8b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/kernelcache.release.iphone6
ipsw/BuildManifest.plist
ipsw/058-74917-062.dmg
ipsw/._058-74917-062.dmg
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
=================================================================

musab7777, 2019-10-29:
Thank you so much sir❤️
I tried with last update you did and now it’s work great without any issue.
iPad mini 2 wifi
ipad4,4
MacOS High Sierra 10.13.6

MattStar45, 2019-11-01:
same issue here, Catalina 10.15.1

ERROR: Unable to connect to device
[=========================================         ] 80.9%Version: 3135d5ce73f7cf7229c373db2f5c333a1c15196b - 359
[TSSC] manually specified ECID to use, parsed "4911464396080" to dec:4911464396080 hex:4778a173530
[TSSC] manually specified ApNonce to use, parsed "228dda81e6298dc0f7cc8c1c4eb5d08709f6b3a0" to hex:228dda81e6298dc0f7cc8c1c4eb5d08709f6b3a0
[TSSC] opening manifests/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

iOS 10.3.3 for device iPhone6,1 IS being signed!
4911464396080_iPhone6,1_10.3.3-14G60_228dda81e6298dc0f7cc8c1c4eb5d08709f6b3a0.shsh -> shsh/apnonce.shsh2
Done prepping files! Time to downgrade!!!
****RESTORING!****
Waiting for device to reconnect...
Version: 536fee9e67dbc2842b2e461bb0d23cfd0f6cf903 - 246
Odysseus support: no
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/apnonce.shsh2 is done
Found device iPhone6,1 n51ap
[TSSC] opening manifests/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Did set SEP+baseband path and firmware
[WARNING] Failed to read BasebandGoldCertID from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandGoldCertID. This might result in invalid baseband signing status information
[WARNING] Failed to read BasebandSerialNumber from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandSerialNumber size. This might result in invalid baseband signing status information
[TSSC] opening manifests/BuildManifest_iPhone6,1_1033_OTA.plist
[TSSR] User specified to request only a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Found device in DFU mode
[Error] unsupported device mode, please put device in recovery mode or normal mode
[Error] Fail code=-3
Failed with errorcode=-3
Cleaning up :D
dummy_file
iBSS.im4p
iBSS.img4
iBSS.iphone6.RELEASE.im4p
iBSS.prepatched
iBSS.raw
iBEC.im4p
iBEC.img4
iBEC.iphone6.RELEASE.im4p
iBEC.prepatched
iBEC.raw
Mav7Mav8-7.60.00.Release.bbfw
sep-firmware.n51.RELEASE.im4p
ipsw/Firmware/aopfw.im4p
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware/dfu/iBEC.n69.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.n69u.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.iphone6.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.n69.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.n69u.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.n69u.RELEASE.im4p
[... 75 lines cut ...]
ipsw/Firmware/all_flash/iBoot.iphone6.RELEASE.im4p
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s8003.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.n53.RELEASE.im4p
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/LLB.n69u.RELEASE.im4p
ipsw/Firmware/all_flash
ipsw/Firmware/Mav10-5.62.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav10-5.62.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/kernelcache.release.iphone8b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/kernelcache.release.iphone6
ipsw/BuildManifest.plist
ipsw/058-74917-062.dmg
ipsw/._058-74917-062.dmg
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

zodaema, 2019-11-09:
You can try old version of this script(step by step).
Full automatic not working.

Yuanlailai, 2019-11-09:
Same problem, iPhone6,2, Mojave 10.14.6

iOS 10.3.3 for device iPhone6,2 IS being signed!
4912515203536_iPhone6,2_10.3.3-14G60_41234383e4311bca58b55807009f68c202d1a4f6.shsh -> shsh/apnonce.shsh2
Done prepping files! Time to downgrade!!!
****RESTORING!****
Waiting for device to reconnect...
Version: 7e66824796f697ea54597dadbdba462382f004b9 - 247
Odysseus support: no
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/apnonce.shsh2 is done
Found device iPhone6,2 n53ap
[TSSC] opening manifests/BuildManifest_iPhone6,2_1033_OTA.plist
[TSSR] User specified not to request a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Did set SEP+baseband path and firmware
[WARNING] Failed to read BasebandGoldCertID from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandGoldCertID. This might result in invalid baseband signing status information
[WARNING] Failed to read BasebandSerialNumber from device! Is it already in recovery?
[WARNING] Using tsschecker's fallback BasebandSerialNumber size. This might result in invalid baseband signing status information
[TSSC] opening manifests/BuildManifest_iPhone6,2_1033_OTA.plist
[TSSR] User specified to request only a baseband ticket.
Request URL set to https://gs.apple.com/TSS/controller?action=2
Sending TSS request attempt 1... response successfully received
Found device in DFU mode
**[Error] unsupported device mode, please put device in recovery mode or normal mode
[Error] Fail code=-3
Failed with errorcode=-3**

_Here the screen should be green.(in the video)_

Cleaning up :D
dummy_file
iBSS.iphone6.RELEASE.im4p
iBEC.iphone6.RELEASE.im4p
Mav7Mav8-7.60.00.Release.bbfw
sep-firmware.n53.RELEASE.im4p
ipsw/Firmware/aopfw.im4p
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware/dfu/iBEC.n69.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.n69u.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.n69.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.n69u.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.n69u.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.iphone6.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.n69.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.iphone6.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.n69u.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.n69.RELEASE.im4p
ipsw/Firmware/dfu
ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s8000.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.n51.RELEASE.im4p
ipsw/Firmware/all_flash/iBoot.n69u.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.n51.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.n69.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s5l8960x.im4p
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8003.im4p
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/recoverymode@1136~iphone-lightning.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s8000.im4p
ipsw/Firmware/all_flash/sep-firmware.n69u.RELEASE.im4p.plist
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s5l8960x.im4p.plist
[... 52 lines cut ...]
ipsw/Firmware/all_flash/iBoot.iphone6.RELEASE.im4p
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s8003.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.n53.RELEASE.im4p
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/LLB.n69u.RELEASE.im4p
ipsw/Firmware/all_flash
ipsw/Firmware/Mav10-5.62.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav10-5.62.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/kernelcache.release.iphone8b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/kernelcache.release.iphone6
ipsw/BuildManifest.plist
ipsw/058-74917-062.dmg
ipsw/._058-74917-062.dmg
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

Littles1225, 2019-11-12:
same issue here,MAC OS 10.14.6

Waiting for device to reconnect...
./restore.sh: line 289: futurerestore: command not found
Cleaning up :D
dummy_file
iBSS.ipad4b.RELEASE.im4p
iBEC.ipad4b.RELEASE.im4p
Mav7Mav8-7.60.00.Release.bbfw
sep-firmware.j85.RELEASE.im4p
ipsw/._058-75094-062.dmg
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu
ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p
ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p
ipsw/Firmware/all_flash/glyphplugin@2x~ipad-lightning.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j86ap.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j86ap.im4p
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j72.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j71ap.im4p
ipsw/Firmware/all_flash/batterylow1@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j72.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j73ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j73.RELEASE.im4p
ipsw/Firmware/all_flash/iBoot.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/batterylow1@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j73ap.im4p
ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j71ap.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4.RELEASE.im4p.plist
ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j73.RELEASE.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p
ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/glyphplugin@2x~ipad-lightning.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/058-75094-062.dmg
ipsw/kernelcache.release.ipad4
ipsw/kernelcache.release.ipad4b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/BuildManifest.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

Merculous, 2019-11-18:
Duplicate of #66

closed 2019-11-18 by Merculous