MatthewPierson/1033-OTA-Downgrader #46: Error in restore.sh   sending files interrupt in 7%

Leadereric, 2019-10-29:
Sending filesystem now...[====                                              ]   7.0%ERROR: Unable to send data to ASR. Sent 0 of 1450 bytes.
ERROR: Unable to send filesystem payload
ERROR: Unable to send payload to ASR
ERROR: Unable to send filesystem
ERROR: Unable to successfully restore device
Got status message
Status: FailASR: __decompressChunk() error: 1000XSTA
ASR: fail
ASR: Could not restore - error 1000
ASR STATUS: fail
restore_apfs_image : failed to restore an APFS image, error = 14
[13:04:40.0806-GMT]{5>7} CHECKPOINT FAILURE:(FAILURE:14) APFS_RELOAD:[0x0C05] restore_invert [0]D(failed to restore APFS image)
restore-step-results = {0x11070C05:{0:14}}
restore-step-codes = {0x11070C05:{0:14}}
restore-step-domains = {0x11070C05:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x11070C05:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 139
restore-step-user-progress = 12
[13:04:40.0810-GMT]{5>7} CHECKPOINT NOTICE: (NVRAM set) restore-step-user-progress=12 [sync=true] (first failure)
[13:04:40.0810-GMT]{5>7} CHECKPOINT FAILURE:(FAILURE:14) RESTORED:[0x064F] check_apfs_format [0]D(failed to restore APFS image)
restore-step-results = {0x1107064F:{0:14};0x11070C05:{0:14}}
restore-step-codes = {0x1107064F:{0:14};0x11070C05:{0:14}}
restore-step-domains = {0x1107064F:{0:"AMRestoreErrorDomain"};0x11070C05:{0:"AMRestoreErrorDomain"}}
restore-step-error = {0x1107064F:"[0]D(failed to restore APFS image)"}
restore-step-uptime = 139
restore-step-user-progress = 12
[13:04:40.0814-GMT]{5>7} CHECKPOINT BEGIN: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103064F:49;0x11030C05:54;0x11030647:55}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert;0x11030647:cleanup_check_result}
restore-step-uptime = 139
restore-step-user-progress = 12
[13:04:40.0817-GMT]{5>7} CHECKPOINT END: RESTORED:[0x0647] cleanup_check_result
restore-step-ids = {0x1103064F:49;0x11030C05:54}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert}
restore-step-uptime = 139
restore-step-user-progress = 12
[13:04:40.0819-GMT]{5>7} CHECKPOINT BEGIN: RESTORED:[0x0648] cleanup_send_final_status
restore-step-ids = {0x1103064F:49;0x11030C05:54;0x11030648:56}
restore-step-names = {0x1103064F:check_apfs_format;0x11030C05:restore_invert;0x11030648:cleanup_send_final_status}
restore-step-uptime = 139
restore-step-user-progress = 12

[Error] ERROR: Unable to restore device
Done: restoring failed.
Failed with errorcode=-11
[Log] Futurerestoring complete

**************** Downgrade complete! Enjoy 10.3.3 =) ****************
**************** Follow me on twitter @mosk_i for help/updates ******
leaders-Mac:1033-OTA-Downgrader-master leader$

Leadereric, 2019-10-29:
My device is iPad4,4

linkway321, 2019-10-29:
My ipad air has been downgraded successfully, ipad mini2 has the same problem as you.

MatthewPierson, 2019-10-29:
@Leadereric Please try again with the latest version of the script

Leadereric, 2019-10-30:
first i have tried the lasted version ,  but it seems to interrupt in command line16 or somthing，just keep showing waiting 10 seconds to put device in dfu mode,but actually my device has been in dfu even pwn mode，yet the command did not start the restore

Merculous, 2019-11-18:
I've never seen this issue before. Could you run everything again and send the full output?