MatthewPierson/1033-OTA-Downgrader #48: Requesting the URL set failed

Leadereric, 2019-10-30:
iBoot64Patcher				ipwndfu_public
Supported Device
iPad4,4
Supported device found.
7926522982792
ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p -> ./iBEC.ipad4b.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p -> ./iBSS.ipad4b.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p -> ./sep-firmware.j85.RELEASE.im4p
img4tool version: 0.152-479a551ea5127e013086aca5e3a6af061f8e2a8a
Extracted IM4P payload to iBEC.raw
main: Starting...
getting get_debug_enabled_patch() patch
getting get_sigcheck_patch() patch
applying patch=0x83000cbfc : 000080d2c0035fd6
applying patch=0x83000de38 : 200080d2
main: Writing out patched file to iBEC.prepatched...
main: Quitting...
img4tool version: 0.152-479a551ea5127e013086aca5e3a6af061f8e2a8a
Created IM4P file at iBEC.im4p
img4tool version: 0.152-479a551ea5127e013086aca5e3a6af061f8e2a8a
Extracted IM4P payload to iBSS.raw
main: Starting...
getting get_sigcheck_patch() patch
applying patch=0x18038bc98 : 000080d2c0035fd6
main: Writing out patched file to iBSS.prepatched...
main: Quitting...
img4tool version: 0.152-479a551ea5127e013086aca5e3a6af061f8e2a8a
Created IM4P file at iBSS.im4p
Version: 3135d5ce73f7cf7229c373db2f5c333a1c15196b - 359
[TSSC] manually specified ECID to use, parsed "7926522982792" to dec:7926522982792 hex:73589963188
[TSSC] opening manifests/BuildManifest_iPad4,4_1033_OTA.plist
[TSSR] LOG: device iPad4,4 doesn't need a baseband ticket, continuing without requesting a Baseband ticket
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] 
[TSSR] Request URL set to https://17.171.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 2... failure
[Error] LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to 17.171.36.30:443 
[TSSR] Request URL set to https://17.151.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 3... failure
[Error] LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to 17.151.36.30:443 
[TSSR] Request URL set to http://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 4... failure
[Error] LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to 17.151.36.30:443 
[TSSR] Request URL set to http://17.171.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 5... failure
[Error] Empty reply from server
[TSSR] Request URL set to http://17.151.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 6... failure
[Error] Empty reply from server
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 7... failure
[Error] Empty reply from server
[TSSR] Request URL set to https://17.171.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 8... failure
[Error] LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to 17.171.36.30:443 
[TSSR] Request URL set to https://17.151.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 9... failure
[Error] LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to 17.151.36.30:443 
[TSSR] Request URL set to http://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 10... failure
[Error] LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to 17.151.36.30:443 
[TSSR] Request URL set to http://17.171.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 11... failure
[Error] Empty reply from server
[TSSR] Request URL set to http://17.151.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 12... failure
[Error] Empty reply from server
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 13... failure
[Error] Empty reply from server
[TSSR] Request URL set to https://17.171.36.30/TSS/controller?action=2
[TSSR] Sending TSS request attempt 14... failure
[Error] LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to 17.171.36.30:443 
[TSSR] Request URL set to https://17.151.36.30/TSS/controller?action=2

DavidKoller2, 2019-10-30:
It maybe a connection problrm

Merculous, 2019-11-18:
This is just due to that WiFi iPad's do not have a baseband. I'll fix this in the next update.