MatthewPierson/1033-OTA-Downgrader #50: ERROR: Exploit failed. Device did not enter pwned DFU Mode.

Ay4y0, 2019-10-30:
Hi bro!
Not one of my iPads is entered in pwndfu mode.

I use:
Macbook A1370 os 10.14 and Macbook A1342 os 10.11
iPad Air 4.1 and iPad Air 4.2

I tried reinstalling 10.14, but it did not bring any results.
I loaded the device into DFU and tried again and again, 10 times, 20 ... 30 ... but always the same "ERROR: Exploit failed. Device did not enter pwned DFU Mode."
I tried on different laptops and different systems, with different iPads, but nothing works.
Please help, I'm already tired of fighting him ((
________________________________________________

`Last login: Tue Oct 29 03:24:52 on console
MacBook-Air-Andy:~ andy$ cd /Users/andy/Desktop/1033-OTA-Downgrader-master 
MacBook-Air-Andy:1033-OTA-Downgrader-master andy$ ./restore.sh
Usage: ./restore.sh PathToIpsw (ipsw must be in this directory)
Example: ./restore.sh iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
MacBook-Air-Andy:1033-OTA-Downgrader-master andy$ ./restore.sh iPad_64bit_10.3.3_14G60_Restore.ipsw 
OOOOF, brew is not installed? Installing...
==> This script will install:
/usr/local/bin/brew
/usr/local/share/doc/homebrew
/usr/local/share/man/man1/brew.1
/usr/local/share/zsh/site-functions/_brew
/usr/local/etc/bash_completion.d/brew
/usr/local/Homebrew
==> The following existing directories will be made group writable:
/usr/local/lib
/usr/local/sbin
==> The following existing directories will have their owner set to andy:
/usr/local/lib
/usr/local/sbin
==> The following existing directories will have their group set to admin:
/usr/local/lib
/usr/local/sbin
==> The following new directories will be created:
/usr/local/bin
/usr/local/etc
/usr/local/include
/usr/local/share
/usr/local/var
/usr/local/opt
/usr/local/share/zsh
/usr/local/share/zsh/site-functions
/usr/local/var/homebrew
/usr/local/var/homebrew/linked
/usr/local/Cellar
/usr/local/Caskroom
/usr/local/Homebrew
/usr/local/Frameworks
==> The Xcode Command Line Tools will be installed.

Press RETURN to continue or any other key to abort
==> /usr/bin/sudo /bin/chmod u+rwx /usr/local/lib /usr/local/sbin
Password:
==> /usr/bin/sudo /bin/chmod g+rwx /usr/local/lib /usr/local/sbin
==> /usr/bin/sudo /usr/sbin/chown andy /usr/local/lib /usr/local/sbin
==> /usr/bin/sudo /usr/bin/chgrp admin /usr/local/lib /usr/local/sbin
==> /usr/bin/sudo /bin/mkdir -p /usr/local/bin /usr/local/etc /usr/local/include /usr/local/share /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Homebrew /usr/local/Frameworks
==> /usr/bin/sudo /bin/chmod g+rwx /usr/local/bin /usr/local/etc /usr/local/include /usr/local/share /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Homebrew /usr/local/Frameworks
==> /usr/bin/sudo /bin/chmod 755 /usr/local/share/zsh /usr/local/share/zsh/site-functions
==> /usr/bin/sudo /usr/sbin/chown andy /usr/local/bin /usr/local/etc /usr/local/include /usr/local/share /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Homebrew /usr/local/Frameworks
==> /usr/bin/sudo /usr/bin/chgrp admin /usr/local/bin /usr/local/etc /usr/local/include /usr/local/share /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Homebrew /usr/local/Frameworks
==> /usr/bin/sudo /bin/mkdir -p /Users/andy/Library/Caches/Homebrew
==> /usr/bin/sudo /bin/chmod g+rwx /Users/andy/Library/Caches/Homebrew
==> /usr/bin/sudo /usr/sbin/chown andy /Users/andy/Library/Caches/Homebrew
==> Searching online for the Command Line Tools
==> /usr/bin/sudo /usr/bin/touch /tmp/.com.apple.dt.CommandLineTools.installondemand.in-progress
==> Installing Command Line Tools (macOS Mojave version 10.14) for Xcode-10.3
[... 327 lines cut ...]
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libplist 2.0.0_1 is already installed and up-to-date
To reinstall 2.0.0_1, run `brew reinstall libplist`
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:10 ECID:000006F48909A908 IBFL:1C SRTG:[iBoot-1704.10]
ERROR: Exploit failed. Device did not enter pwned DFU Mode.
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode`

Ay4y0, 2019-10-31:
What should I do?((

chenyinfu, 2019-11-01:
same problem

chenyinfu, 2019-11-01:
Traceback (most recent call last):
  File "./ipwndfu", line 66, in <module>
    checkm8.exploit()
  File "/Users/xingzhe/Downloads/1033-OTA-Downgrader-master/ipwndfu_public/checkm8.py", line 533, in exploit
    libusb1_async_ctrl_transfer(device, 0x21, 1, 0, 0, 'A' * 0x800, 0.0001)
  File "/Users/xingzhe/Downloads/1033-OTA-Downgrader-master/ipwndfu_public/checkm8.py", line 45, in libusb1_async_ctrl_transfer
    assert usb.backend.libusb1._lib.libusb_cancel_transfer(transfer_ptr) == 0
AssertionError
./restore.sh: line 304: 37295 Segmentation fault: 11  ./ipwndfu -p

chenyinfu, 2019-11-01:
Process:               Python [37295]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.16 (2.7.16)
Build Info:            python-125000000000000~400
Code Type:             X86-64 (Native)
Parent Process:        bash [34375]
Responsible:           Terminal [456]
User ID:               501

PlugIn Path:             /Users/USER/Downloads/*/libusb-1.0.22.mojave.bottle.dylib
PlugIn Identifier:       libusb-1.0.22.mojave.bottle.dylib
PlugIn Version:          ??? (0)

Date/Time:             2019-11-01 14:47:49.030 +0800
OS Version:            Mac OS X 10.15 (19A583)
Report Version:        12
Anonymous UUID:        39B3C06E-C54E-C577-A22B-368C4D9BEEC0

Time Awake Since Boot: 1700 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [37295]

VM Regions Near 0x58:
--> 
    __TEXT                 0000000107fb4000-0000000107fb6000 [    8K] r-x/r-x SM=COW  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libusb-1.0.22.mojave.bottle.dylib	0x000000010858b5bb remove_from_flying_list + 17
1   libusb-1.0.22.mojave.bottle.dylib	0x000000010858b707 usbi_handle_transfer_completion + 39
2   libusb-1.0.22.mojave.bottle.dylib	0x000000010858c3d9 handle_events + 1012
3   libusb-1.0.22.mojave.bottle.dylib	0x000000010858bd7b libusb_handle_events_timeout_completed + 262
4   libusb-1.0.22.mojave.bottle.dylib	0x00000001085888d5 libusb_exit + 334
5   libffi.dylib                  	0x00007fff6a951915 ffi_call_unix64 + 85
6   libffi.dylib                  	0x00007fff6a951255 ffi_call_int + 699
7   _ctypes.so                    	0x000000010837b650 _ctypes_callproc + 657
8   _ctypes.so                    	0x0000000108375b77 0x108373000 + 11127
9   org.python.python             	0x00007fff409f4319 PyObject_Call + 97
10  org.python.python             	0x00007fff40a7040a PyEval_EvalFrameEx + 20636
11  org.python.python             	0x00007fff40a74a37 0x7fff409ea000 + 567863
12  org.python.python             	0x00007fff40a702c2 PyEval_EvalFrameEx + 20308
13  org.python.python             	0x00007fff40a74a37 0x7fff409ea000 + 567863
14  org.python.python             	0x00007fff40a702c2 PyEval_EvalFrameEx + 20308
15  org.python.python             	0x00007fff40a74a37 0x7fff409ea000 + 567863
16  org.python.python             	0x00007fff40a702c2 PyEval_EvalFrameEx + 20308
17  org.python.python             	0x00007fff40a6ad5e PyEval_EvalCodeEx + 534
18  org.python.python             	0x00007fff40a11c85 0x7fff409ea000 + 162949
19  org.python.python             	0x00007fff409f4319 PyObject_Call + 97
20  org.python.python             	0x00007fff409feafe 0x7fff409ea000 + 84734
21  org.python.python             	0x00007fff409f4319 PyObject_Call + 97
22  org.python.python             	0x00007fff40a743d9 PyEval_CallObjectWithKeywords + 159
23  org.python.python             	0x00007fff40a3a7b7 0x7fff409ea000 + 329655
24  org.python.python             	0x00007fff40a3eba2 0x7fff409ea000 + 347042
25  org.python.python             	0x00007fff40a1f07c 0x7fff409ea000 + 217212
26  org.python.python             	0x00007fff40a3ed34 0x7fff409ea000 + 347444
27  org.python.python             	0x00007fff40a1f07c 0x7fff409ea000 + 217212
28  org.python.python             	0x00007fff40a3ed34 0x7fff409ea000 + 347444
29  org.python.python             	0x00007fff40a20356 0x7fff409ea000 + 222038
30  org.python.python             	0x00007fff40a1ec57 0x7fff409ea000 + 216151
31  org.python.python             	0x00007fff40a1e345 0x7fff409ea000 + 213829
[... 144 lines cut ...]
__OBJC_RW                         1764K        2 
__TEXT                            15.2M       81 
__UNICODE                          564K        1 
mapped file                         28K        1 
shared memory                       12K        3 
===========                     =======  ======= 
TOTAL                            514.4M      271 

Model: MacBookAir5,2, BootROM 260.0.0.0.0, 2 processors, Dual-Core Intel Core i3, 2.49 GHz, 4 GB, SMC 2.5f9
Graphics: kHW_IntelHD4000Item, Intel HD Graphics 4000, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, SK Hynix, HMT325S6CFR8C-PBN0
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, SK Hynix, HMT325S6CFR8C-PBN0
Network Service: Ethernet, Ethernet, en0
PCI Card: sppci_expresscard_name, sppci_unknown
Serial ATA Device: Teclast 128GB S550, 128.04 GB
Serial ATA Device: HGST HTS725050A7E630, 500.11 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Integrated Camera
USB Device: Apple Mobile Device (DFU Mode)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: ThinkPad USB Laser Mouse
USB Device: USB 3.0 Bus
Thunderbolt Bus:

Ay4y0, 2019-11-04:
I wonder why this section on the site is needed if no one is commenting and helping to solve the problem. Or maybe it depends on the author?...

Deppjones123, 2019-11-06:
iOS 12.4 made my iPad Air 1 so slow, that I would like to go back to 10.3.3 but I have the same problem. My iPad doesn't enter the PWNDFU-Mode. Using a MacPro with Mojave to downgrade.

Phaptlq, 2019-11-14:
i have same problem.

Merculous, 2019-11-18:
Homebrew refuses to run anything as root. As for the actual issue, this may be pyusb related.