MatthewPierson/1033-OTA-Downgrader #51: hi bro,

VanessaC555, 2019-10-31:
Error: Running Homebrew as root is extremely dangerous and no longer supported.
As Homebrew does not drop privileges on installation you would be giving all
build scripts full access to your system.

MatthewPierson, 2019-10-31:
Don't run the script as root. I never say in the readme to do so

VanessaC555, 2019-10-31:
> Don't run the script as root. I never say in the readme to do so

Well, what does the link blow mean??
https://github.com/MatthewPierson/1033-OTA-Downgrader/issues/50#issuecomment-548020832

MatthewPierson, 2019-10-31:
Ignore them. They aren't me so don't do things they say to and expect it to work.

VanessaC555, 2019-10-31:
All right!
So, how to solve the problem the link describes ?  I have the same problem.
https://github.com/MatthewPierson/1033-OTA-Downgrader/issues/50#issue-514600690

Merculous, 2019-11-18:
This may be a pyusb related error. I am going to look into this further.