MatthewPierson/1033-OTA-Downgrader #53: raise ValueError("The device has no langid")

kaiseryeahhh, 2019-10-31:
raise ValueError("The device has no langid")
ValueError: The device has no langid

Fail in catalina and high sierra
iPad mini 2

Alex-J-Fox, 2019-10-31:
I'm having the same issue. 

`Traceback (most recent call last):
  File "./ipwndfu", line 48, in <module>
    serial_number = device.serial_number
  File "/Users/alexfox/Desktop/1033-OTA-Downgrader-master/ipwndfu_public/usb/core.py", line 830, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)
  File "/Users/alexfox/Desktop/1033-OTA-Downgrader-master/ipwndfu_public/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid`

macOS 10.14.6
iPad mini 2

MattStar45, 2019-11-01:
what I ddi to fix that was I put the device in dfu mode again, try that. if it doesn't work, then idk

MatthewPierson, 2019-11-13:
"The device has no langid" is an issue with ipwndfu, not my script specifically. Google around for the fix for that, there is one but can't remember it off the top of my head

Merculous, 2019-11-18:
Normal issue with ipwndfu. Rebooting and trying again will fix this.

closed 2019-11-18 by Merculous