MatthewPierson/1033-OTA-Downgrader #56: while use plugin "libusb-1.0.22.mojave.bottle.dylib" Python exit?

chenyinfu, 2019-11-04:
Traceback (most recent call last):
  File "./ipwndfu", line 66, in <module>
    checkm8.exploit()
  File "/Users/xingzhe/Downloads/1033-OTA/ipwndfu_public/checkm8.py", line 533, in exploit
    libusb1_async_ctrl_transfer(device, 0x21, 1, 0, 0, 'A' * 0x800, 0.0001)
  File "/Users/xingzhe/Downloads/1033-OTA/ipwndfu_public/checkm8.py", line 45, in libusb1_async_ctrl_transfer
    assert usb.backend.libusb1._lib.libusb_cancel_transfer(transfer_ptr) == 0
AssertionError
./restore.sh: line 304: 17303 Segmentation fault: 11  ./ipwndfu -p
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode

chenyinfu, 2019-11-04:
Process:               Python [17303]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.16 (2.7.16)
Build Info:            python-125000000000000~400
Code Type:             X86-64 (Native)
Parent Process:        bash [9671]
Responsible:           Terminal [807]
User ID:               501

PlugIn Path:             /Users/USER/Downloads/*/libusb-1.0.22.mojave.bottle.dylib
PlugIn Identifier:       libusb-1.0.22.mojave.bottle.dylib
PlugIn Version:          ??? (0)

Date/Time:             2019-11-04 09:35:12.462 +0800
OS Version:            Mac OS X 10.15 (19A583)
Report Version:        12
Anonymous UUID:        39B3C06E-C54E-C577-A22B-368C4D9BEEC0

Time Awake Since Boot: 3600 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [17303]

VM Regions Near 0x58:
--> 
    __TEXT                 000000010ed9b000-000000010ed9d000 [    8K] r-x/r-x SM=COW  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libusb-1.0.22.mojave.bottle.dylib	0x000000010f3725bb remove_from_flying_list + 17
1   libusb-1.0.22.mojave.bottle.dylib	0x000000010f372707 usbi_handle_transfer_completion + 39
2   libusb-1.0.22.mojave.bottle.dylib	0x000000010f3733d9 handle_events + 1012
3   libusb-1.0.22.mojave.bottle.dylib	0x000000010f372d7b libusb_handle_events_timeout_completed + 262
4   libusb-1.0.22.mojave.bottle.dylib	0x000000010f36f8d5 libusb_exit + 334
5   libffi.dylib                  	0x00007fff65b82915 ffi_call_unix64 + 85
6   libffi.dylib                  	0x00007fff65b82255 ffi_call_int + 699
7   _ctypes.so                    	0x000000010f162650 _ctypes_callproc + 657
8   _ctypes.so                    	0x000000010f15cb77 0x10f15a000 + 11127
9   org.python.python             	0x00007fff3bc25319 PyObject_Call + 97
10  org.python.python             	0x00007fff3bca140a PyEval_EvalFrameEx + 20636
11  org.python.python             	0x00007fff3bca5a37 0x7fff3bc1b000 + 567863
12  org.python.python             	0x00007fff3bca12c2 PyEval_EvalFrameEx + 20308
13  org.python.python             	0x00007fff3bca5a37 0x7fff3bc1b000 + 567863
14  org.python.python             	0x00007fff3bca12c2 PyEval_EvalFrameEx + 20308
15  org.python.python             	0x00007fff3bca5a37 0x7fff3bc1b000 + 567863
16  org.python.python             	0x00007fff3bca12c2 PyEval_EvalFrameEx + 20308
17  org.python.python             	0x00007fff3bc9bd5e PyEval_EvalCodeEx + 534
18  org.python.python             	0x00007fff3bc42c85 0x7fff3bc1b000 + 162949
19  org.python.python             	0x00007fff3bc25319 PyObject_Call + 97
20  org.python.python             	0x00007fff3bc2fafe 0x7fff3bc1b000 + 84734
21  org.python.python             	0x00007fff3bc25319 PyObject_Call + 97
22  org.python.python             	0x00007fff3bca53d9 PyEval_CallObjectWithKeywords + 159
23  org.python.python             	0x00007fff3bc6b7b7 0x7fff3bc1b000 + 329655
24  org.python.python             	0x00007fff3bc6fba2 0x7fff3bc1b000 + 347042
25  org.python.python             	0x00007fff3bc5007c 0x7fff3bc1b000 + 217212
26  org.python.python             	0x00007fff3bc6fd34 0x7fff3bc1b000 + 347444
27  org.python.python             	0x00007fff3bc5007c 0x7fff3bc1b000 + 217212
28  org.python.python             	0x00007fff3bc6fd34 0x7fff3bc1b000 + 347444
29  org.python.python             	0x00007fff3bc51356 0x7fff3bc1b000 + 222038
30  org.python.python             	0x00007fff3bc4fc57 0x7fff3bc1b000 + 216151
31  org.python.python             	0x00007fff3bc4f345 0x7fff3bc1b000 + 213829
[... 143 lines cut ...]
__OBJC_RO                         31.8M        1 
__OBJC_RW                         1764K        2 
__TEXT                            15.2M       81 
__UNICODE                          564K        1 
mapped file                         28K        1 
shared memory                       12K        3 
===========                     =======  ======= 
TOTAL                            514.4M      270 

Model: MacBookAir5,2, BootROM 260.0.0.0.0, 2 processors, Dual-Core Intel Core i3, 2.49 GHz, 4 GB, SMC 2.5f9
Graphics: kHW_IntelHD4000Item, Intel HD Graphics 4000, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, SK Hynix, HMT325S6CFR8C-PBN0
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, SK Hynix, HMT325S6CFR8C-PBN0
Network Service: Ethernet, Ethernet, en0
PCI Card: sppci_expresscard_name, sppci_unknown
Serial ATA Device: Teclast 128GB S550, 128.04 GB
Serial ATA Device: HGST HTS725050A7E630, 500.11 GB
USB Device: USB 3.0 Bus
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: ThinkPad USB Laser Mouse
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Integrated Camera
Thunderbolt Bus:

chenyinfu, 2019-11-04:
I try reinstall homebrew,but same problem.

csyhoyang, 2019-12-04:
我也遇到同样的问题

frank0767, 2020-01-18:
Same problem

paquoc, 2020-03-20:
same problem too :'(