MatthewPierson/1033-OTA-Downgrader #58: ./restore.sh: line 289: futurerestore: command not found

R1ehen, 2019-11-05:
What  this problem?
Killing iTunes as this will be quite annoying with what we are going to do.
No matching processes belonging to you were found
Archive:  /Volumes/Mac/iPad_64bit_10.3.3_14G60_Restore.ipsw
  inflating: ipsw/058-74940-063.dmg  
  inflating: ipsw/058-75094-062.dmg  
  inflating: ipsw/._058-75094-062.dmg  
  inflating: ipsw/058-75381-062.dmg  
  inflating: ipsw/BuildManifest.plist  
   creating: ipsw/Firmware/
   creating: ipsw/Firmware/all_flash/
  inflating: ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterylow1@2x~ipad.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterylow1@2x~ipad.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j71ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j71ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j73ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j73ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j86ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j86ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/glyphplugin@2x~ipad-lightning.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/glyphplugin@2x~ipad-lightning.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/iBoot.ipad4.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/iBoot.ipad4.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/recoverymode@2x~ipad-lightning.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j71.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j72.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j72.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j73.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j73.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p  
  inflating: ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p.plist  
   creating: ipsw/Firmware/dfu/
  inflating: ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p  
  inflating: ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p  
  inflating: ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p  
  inflating: ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p  
  inflating: ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p.plist  
  inflating: ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw  
  inflating: ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist  
[... 199 lines cut ...]
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/058-75094-062.dmg
ipsw/kernelcache.release.ipad4
ipsw/kernelcache.release.ipad4b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/BuildManifest.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
riehen@wenghongdeMacBook-Air Mac %

MatthewPierson, 2019-11-05:
You need to show me the entire log for me to be able to do anything

R1ehen, 2019-11-06:
ok,this's entire log in here.

riehen@wenghongdeMacBook-Air ~ % cd /Volumes/Mac
riehen@wenghongdeMacBook-Air Mac % ./restore.sh /Volumes/Mac/iPad_64bit_10.3.3_14G60_Restore.ipsw
[+] Installing dependencies
Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libplist 2.0.0_1 is already installed and up-to-date
To reinstall 2.0.0_1, run `brew reinstall libplist`
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:0A ECID:000005350868BD48 IBFL:1C SRTG:[iBoot-1704.10]
Device is now in pwned DFU Mode.
(27.41 seconds)
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:0A ECID:000005350868BD48 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Traceback (most recent call last):
  File "rmsigchks.py", line 114, in <module>
    main()
  File "rmsigchks.py", line 96, in main
    pdev = usbexec.PwnedUSBDevice()
  File "/Volumes/Mac/ipwndfu_public/usbexec.py", line 135, in __init__
    self.serial_number = device.serial_number
  File "/Volumes/Mac/ipwndfu_public/usb/core.py", line 830, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)
  File "/Volumes/Mac/ipwndfu_public/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
We seem to be in pwned DFU mode!
[+] Build folder exists! Updating only!
From https://github.com/merculous/futurerestore
 * branch            master     -> FETCH_HEAD
Already up to date.
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 8 (delta 1), reused 3 (delta 0), pack-reused 0
Unpacking objects: 100% (8/8), done.
From https://github.com/tihmstar/img4tool
 * branch            master     -> FETCH_HEAD
   7186b00..d514ce5  master     -> origin/master
Updating 7186b00..d514ce5
Fast-forward
 img4tool/img4tool.cpp            | 23 +++++++++++++++++++++++
 img4tool/main.cpp                |  9 ++++++++-
 include/img4tool/img4tool.hpp.in |  1 +
 3 files changed, 32 insertions(+), 1 deletion(-)
From https://github.com/tihmstar/libfragmentzip
 * branch            master     -> FETCH_HEAD
Already up to date.
[... 296 lines cut ...]
ipsw/Firmware/all_flash/DeviceTree.j87ap.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist
ipsw/Firmware/all_flash/iBoot.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.ipad4.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~ipad.s5l8960x.im4p
ipsw/Firmware/all_flash/DeviceTree.j85ap.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j85.RELEASE.im4p
ipsw/Firmware/all_flash/DeviceTree.j72ap.im4p
ipsw/Firmware/all_flash/applelogo@2x~ipad.s5l8960x.im4p.plist
ipsw/Firmware/all_flash
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/058-75094-062.dmg
ipsw/kernelcache.release.ipad4
ipsw/kernelcache.release.ipad4b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/BuildManifest.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
riehen@wenghongdeMacBook-Air Mac %

MatthewPierson, 2019-11-06:
Can you delete the "build" folder and re run the script? Looks like something failed the first time you ran it when it tried to build something. Deleting the build folder and re-running the script should fix the error. Please show me the log again if it doesn't though

R1ehen, 2019-11-07:
I have already degraded 10.3.3. I succeeded only by using StableA7.sh
The script error code -3.So I think this problem still exists.
Thank you for your help.

closed 2019-11-08 by R1ehen

Littles1225, 2019-11-13:
> I have already degraded 10.3.3. I succeeded only by using StableA7.sh
> The script error code -3.So I think this problem still exists.
> Thank you for your help.

Hi,guys, i have the some issue,can u tell how to do,tks

R1ehen, 2019-11-13:
> > I have already degraded 10.3.3. I succeeded only by using StableA7.sh
> > The script error code -3.So I think this problem still exists.
> > Thank you for your help.
> 
> Hi,guys, i have the some issue,can u tell how to do,tks

You can try StableA7 scripts. Run ./StableA7.sh in terminal

Littles1225, 2019-11-13:
> > > I have already degraded 10.3.3. I succeeded only by using StableA7.sh
> > > The script error code -3.So I think this problem still exists.
> > > Thank you for your help.
> > 
> > 
> > Hi,guys, i have the some issue,can u tell how to do,tks
> 
> You can try StableA7 scripts. Run ./StableA7.sh in terminal

ok,tks

Littles1225, 2019-11-13:
> > > I have already degraded 10.3.3. I succeeded only by using StableA7.sh
> > > The script error code -3.So I think this problem still exists.
> > > Thank you for your help.
> > 
> > 
> > Hi,guys, i have the some issue,can u tell how to do,tks
> 
> You can try StableA7 scripts. Run ./StableA7.sh in terminal

Have trid，But still failed:

Sending NORData now...
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Checking filesystems (15)
Checking filesystems (15)
Checking filesystems (15)
Unmounting filesystems (29)
Unmounting filesystems (29)
Creating partition map (11)
Creating filesystem (12)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
[==================================================]  99.0%
==> Restoring failed. Attempting to exit recovery mode...
./StableA7.sh: line 352: 78322 Abort trap: 6           bin/futurerestore --exit-recovery >&/dev/null

MatthewPierson, 2019-11-13:
Please don't comment about the other script here. That is not relevant to my stuff at all, if you have an issue with it then ask the creator for help

Littles1225, 2019-11-13:
> Please don't comment about the other script here. That is not relevant to my stuff at all, if you have an issue with it then ask the creator for help

i'm so sorry about that.and  i use ./restore.sh，then have the issure:
waiting for device to reconnect...
./restore.sh:line 289:futurerestore:command not found
Cleaning up :D

how to solve,help me,please