MatthewPierson/1033-OTA-Downgrader #59: iPhone 6,1 auto restart to normal mode when process.

zodaema, 2019-11-08:
[image]

Always restart when in the process as picture.

zodaema, 2019-11-08:
[image]

MatthewPierson, 2019-11-08:
You need to show me the entire log, not just small screen shots, if you want me to help

zodaema, 2019-11-08:
> You need to show me the entire log, not just small screen shots, if you want me to help

`Last login: Fri Nov  8 14:55:44 on console
Nattaphons-MacBook-Pro:~ nattaphonsuparsri$ cd
Nattaphons-MacBook-Pro:~ nattaphonsuparsri$ cd /Users/nattaphonsuparsri/Desktop/1033-OTA-Downgrader-master 
Nattaphons-MacBook-Pro:1033-OTA-Downgrader-master nattaphonsuparsri$ ./restore.sh /Users/nattaphonsuparsri/Desktop/1033-OTA-Downgrader-master/iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw 
[+] Installing dependencies
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
cheat        kops         lerna        libssh       platformio   vault

Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libplist 2.0.0_1 is already installed and up-to-date
To reinstall 2.0.0_1, run `brew reinstall libplist`
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:00 ECID:00000677F2CEDA50 IBFL:1C SRTG:[iBoot-1704.10]
Device is now in pwned DFU Mode.
(12.92 seconds)
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:00 ECID:00000677F2CEDA50 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Successfully applied patches
Resetting device state
* This will effectiveley disable pwned DFU Mode
* Only the signature patches will remain
Device is now ready to accept unsigned images
We seem to be in pwned DFU mode!
[+] Build folder exists! Updating only!
From https://github.com/merculous/futurerestore
 * branch            master     -> FETCH_HEAD
Already up to date.
^@From https://github.com/tihmstar/img4tool
 * branch            master     -> FETCH_HEAD
Already up to date.
^R
From https://github.com/tihmstar/libfragmentzip
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/libimobiledevice/libimobiledevice
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/libimobiledevice/libirecovery
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/tihmstar/liboffsetfinder64
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/libimobiledevice/libusbmuxd
[... 381 lines cut ...]
ipsw/Firmware/dfu/iBEC.n69.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.n69u.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.n69u.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.iphone6.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.n69.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.n69.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.n69u.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.n69u.RELEASE.im4p.plist
ipsw/Firmware/dfu
ipsw/Firmware/Mav10-5.62.00.Release.bbfw
ipsw/Firmware/Mav10-5.62.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware
ipsw/kernelcache.release.iphone6
ipsw/kernelcache.release.iphone8b
ipsw/Restore.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
Nattaphons-MacBook-Pro:1033-OTA-Downgrader-master nattaphonsuparsri$ 
`

MatthewPierson, 2019-11-08:
Can you delete the "build" folder that was created and try again? Then show me the full log of that as well

zodaema, 2019-11-08:
[image]

I've trying your old commit.

Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (653333 bytes)...
ERROR: Unable to send iBEC component: Unable to find device
ERROR: Unable to send iBEC to device.
[Error] ERROR: Unable to send iBEC

same error code = -8

zodaema, 2019-11-08:
Try again, and no save file.

It success!!!

Many thanks

[image]

closed 2019-11-08 by zodaema