MatthewPierson/1033-OTA-Downgrader #60: FutureRestore: command not found

ThisIsNoahEvans, 2019-11-09:
Hi,

When downgrading my 5s (iPhone 6,2), it seems to go well until the end. I'm running Mojave on a 2018 MacBook Pro.

```
MacBook-Pro:~ thisisnoahevans$ cd /Users/thisisnoahevans/Local\ Documents/1033-OTA-Downgrader-master 
MacBook-Pro:1033-OTA-Downgrader-master thisisnoahevans$ ./restore.sh /Users/thisisnoahevans/Local\ Documents/1033-OTA-Downgrader-master/iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw 
[+] Installing dependencies
Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libplist 2.0.0_1 is already installed and up-to-date
To reinstall 2.0.0_1, run `brew reinstall libplist`
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
2019-11-09 12:41:35.509 system_profiler[20686:333328] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-09 12:41:35.509 system_profiler[20686:333328] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-09 12:41:35.510 system_profiler[20686:333328] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-09 12:41:35.510 system_profiler[20686:333328] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-09 12:41:35.511 system_profiler[20686:333328] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-09 12:41:35.511 system_profiler[20686:333328] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:000007F4AB50AB30 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Patches have already been applied. Exiting.
We seem to be in pwned DFU mode!
[+] Build folder does not exist! Grabbing dependencies and installing!
Cloning into 'libusbmuxd'...
remote: Enumerating objects: 12, done.
remote: Counting objects: 100% (12/12), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 561 (delta 2), reused 6 (delta 1), pack-reused 549
Receiving objects: 100% (561/561), 190.99 KiB | 873.00 KiB/s, done.
Resolving deltas: 100% (308/308), done.
Cloning into 'libirecovery'...
remote: Enumerating objects: 65, done.
remote: Counting objects: 100% (65/65), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 1154 (delta 32), reused 50 (delta 20), pack-reused 1089
Receiving objects: 100% (1154/1154), 479.57 KiB | 1.47 MiB/s, done.
Resolving deltas: 100% (564/564), done.
Cloning into 'libimobiledevice'...
remote: Enumerating objects: 50, done.
remote: Counting objects: 100% (50/50), done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 8586 (delta 25), reused 31 (delta 15), pack-reused 8536
Receiving objects: 100% (8586/8586), 2.66 MiB | 3.16 MiB/s, done.
Resolving deltas: 100% (6052/6052), done.
Cloning into 'libfragmentzip'...
remote: Enumerating objects: 49, done.
remote: Counting objects: 100% (49/49), done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 355 (delta 22), reused 37 (delta 15), pack-reused 306
Receiving objects: 100% (355/355), 257.09 KiB | 953.00 KiB/s, done.
Resolving deltas: 100% (140/140), done.
Cloning into 'futurerestore'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 1070 (delta 0), reused 1 (delta 0), pack-reused 1066
[... 1417 lines cut ...]
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for update installing
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] faild to build tssrequest for alternative installType
[TSSR] User specified not to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved signing tickets!

iOS 10.3.3 for device iPhone6,2 IS being signed!
8747427605296_iPhone6,2_10.3.3-14G60_e1bbe014306e77fb2a0e204d1faa59564fcee53a.shsh -> shsh/apnonce.shsh2
Done prepping files! Time to downgrade!!!
****RESTORING!****
Waiting for device to reconnect...
./restore.sh: line 285: futurerestore: command not found
Cleaning up :D
dummy_file
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
```

**Thank you in advance!**

MatthewPierson, 2019-11-13:
Please delete the current files and try again with the latest update that I just posted. It should fix this issue + other smaller things

merzota, 2020-01-21:
The same trouble. I used the latest version(

merzota, 2020-01-21:
> Please delete the current files and try again with the latest update that I just posted. It should fix this issue + other smaller things

The same trouble. I used the latest version(

MatthewPierson, 2020-01-21:
Don't use this, use [Vieux](https://github.com/MatthewPierson/Vieux) instead.

closed 2020-01-21 by MatthewPierson