MatthewPierson/1033-OTA-Downgrader #61: what to do next?

yohnz, 2019-11-09:
10.14
```
./restore.sh iPad_64bit_10.3.3_14G60_Restore.ipsw                                                                                                                         !11987
[+] Installing dependencies
Warning: libtool 2.4.6_1 is already installed, it's just not linked
You can use `brew link libtool` to link this version.
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libplist 2.0.0_1 is already installed, it's just not linked
You can use `brew link libplist` to link this version.
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:0A ECID:00000574C94696B0 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Patches have already been applied. Exiting.
We seem to be in pwned DFU mode!
[+] Build folder exists! Updating only!
From https://github.com/merculous/futurerestore
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/tihmstar/img4tool
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/tihmstar/libfragmentzip
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/libimobiledevice/libimobiledevice
 * branch            master     -> FETCH_HEAD
Already up to date.
From https://github.com/libimobiledevice/libirecovery
 * branch            master     -> FETCH_HEAD
Already up to date.
remote: Enumerating objects: 19, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 10 (delta 7), reused 10 (delta 7), pack-reused 0
Unpacking objects: 100% (10/10), done.
From https://github.com/tihmstar/liboffsetfinder64
 * branch            master     -> FETCH_HEAD
   4815b26..c33cf76  master     -> origin/master
Updating 4815b26..c33cf76
Fast-forward
 include/liboffsetfinder64/ibootpatchfinder64.hpp |  35 ++++++++++++++++++++++++-
 liboffsetfinder64.xcodeproj/project.pbxproj      |  26 +++++--------------
 liboffsetfinder64/ibootpatchfinder64.cpp         | 173 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++------------------------------
 liboffsetfinder64/main.cpp                       |   2 +-
 4 files changed, 173 insertions(+), 63 deletions(-)
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (1/1), done.
remote: Total 4 (delta 3), reused 4 (delta 3), pack-reused 0
Unpacking objects: 100% (4/4), done.
From https://github.com/libimobiledevice/libusbmuxd
 * branch            master     -> FETCH_HEAD
   1e7711d..53e832a  master     -> origin/master
Updating 1e7711d..53e832a
Fast-forward
 include/usbmuxd.h | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)
remote: Enumerating objects: 1, done.
[... 251 lines cut ...]
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j86.RELEASE.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.j87.RELEASE.im4p.plist
ipsw/Firmware/all_flash
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.ipad4.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.ipad4b.RELEASE.im4p.plist
ipsw/Firmware/dfu
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware
ipsw/kernelcache.release.ipad4
ipsw/kernelcache.release.ipad4b
ipsw/Restore.plist
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
```

MatthewPierson, 2019-11-13:
Please delete the current files and try again with the latest update that I just posted. It should fix this issue + other smaller things

yohnz, 2019-12-02:
Looks like the Xcode is too outdated.
```
Error: No available formula with the name "libusbmuxd"
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.
Error: No such keg: /usr/local/Cellar/libusbmuxd
Error: Your Xcode (10.1) is too outdated.
Please update to Xcode 10.2.1 (or delete it).
Xcode can be updated from the App Store.

Error: No such keg: /usr/local/Cellar/libimobiledevice
Error: Your Xcode (10.1) is too outdated.
Please update to Xcode 10.2.1 (or delete it).
Xcode can be updated from the App Store.
```

yohnz, 2019-12-15:
Successed on 10.14.6.

yohnz, 2019-12-15:
> Please delete the current files and try again with the latest update that I just posted. It should fix this issue + other smaller things

Success on 10.14.6.

LifeIsBinary, 2019-12-31:
@ I have the same enviroment and device（iPad mini2) ，but use the latest update facing the same condiction(the same error code). can you give me some details which is diffrent to your first work .