MatthewPierson/1033-OTA-Downgrader #66: script error

cryptlord4, 2019-11-10:
hi bro, script error 
how can i do?
need help .... T-T

use
iPhone 6,2 (global)
Mac OS catalina

--------------------------------------------------------------
Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libplist 2.0.0_1 is already installed and up-to-date
To reinstall 2.0.0_1, run `brew reinstall libplist`
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
2019-11-10 22:22:44.977 system_profiler[14869:738506] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:00000077F4411794 IBFL:1C SRTG:[iBoot-1704.10]
Traceback (most recent call last):
  File "./ipwndfu", line 66, in <module>
    checkm8.exploit()
  File "/Users/seokcheol/Downloads/1033-OTA-Downgrader-master/ipwndfu_public/checkm8.py", line 533, in exploit
    libusb1_async_ctrl_transfer(device, 0x21, 1, 0, 0, 'A' * 0x800, 0.0001)
  File "/Users/seokcheol/Downloads/1033-OTA-Downgrader-master/ipwndfu_public/checkm8.py", line 45, in libusb1_async_ctrl_transfer
    assert usb.backend.libusb1._lib.libusb_cancel_transfer(transfer_ptr) == 0
AssertionError
./restore.sh: line 304: 15531 Segmentation fault: 11  ./ipwndfu -p
2019-11-10 22:23:09.497 system_profiler[15537:739643] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:00000077F4411794 IBFL:1C SRTG:[iBoot-1704.10]
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
2019-11-10 22:23:25.942 system_profiler[16207:740920] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
2019-11-10 22:23:42.001 system_profiler[16778:741976] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
[... 579 lines cut ...]
ipsw/Firmware/all_flash/iBoot.iphone6.RELEASE.im4p
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s8003.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.n53.RELEASE.im4p
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/LLB.n69u.RELEASE.im4p
ipsw/Firmware/all_flash
ipsw/Firmware/Mav10-5.62.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav10-5.62.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/kernelcache.release.iphone8b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/kernelcache.release.iphone6
ipsw/BuildManifest.plist
ipsw/058-74917-062.dmg
ipsw/._058-74917-062.dmg
ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

MatthewPierson, 2019-11-13:
Please delete the current files and try again with the latest update that I just posted. It should fix this issue + other smaller things

Merculous, 2019-11-18:
iPhone6,2 isn't supported... Closing..

Merculous, 2019-11-18:
Oops, had a brain fart and thought 6,2 was an iPhone 6.

Merculous, 2019-11-18:
I believe "[Error] Fail code=-3 Failed with errorcode=-3" is the real error.

closed 2019-11-18 by cryptlord4