MatthewPierson/1033-OTA-Downgrader #68: Cannot detect the device in DFU mode

itzxeon78, 2019-11-12:
The script does not detect the device.

I'm on macOS 10.15.1 Catalina.

Logs:

`Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
Traceback (most recent call last):
  File "./ipwndfu", line 47, in <module>
    device = dfu.acquire_device()
  File "/Users/jordiyu/Downloads/1033-OTA-Downgrader-master/ipwndfu_public/dfu.py", line 16, in acquire_device
    for device in usb.core.find(find_all=True, idVendor=0x5AC, idProduct=0x1227, backend=backend):
  File "/Users/jordiyu/Downloads/1033-OTA-Downgrader-master/ipwndfu_public/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
2019-11-12 21:40:42.400 system_profiler[99607:186537] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes were found
Waiting 10 seconds to allow you to enter DFU mode`

Phaptlq, 2019-11-14:
i have same problem. iphone 5s with mac mojave.

Merculous, 2019-11-18:
I could be that pyusb is missing? Please run pip install pysub or python -m easy_install pyusb. If pysub is missing, install that with the command given and please comment if anything has changed.