MatthewPierson/1033-OTA-Downgrader #72: ERROR: Unable to receive message from FDR 0x7fcaf6dd7920 (-2). 0/2 bytes

Netfliics, 2019-11-14:
my device goes to recovery mode during the restore
this is the log:

Air-di-Riccardo:~ riccardocardarelli$ cd /Users/riccardocardarelli/Desktop/1033-OTA-Downgrader-master 
Air-di-Riccardo:1033-OTA-Downgrader-master riccardocardarelli$ ./restore.sh iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
[+] Installing dependencies
Updating Homebrew...
Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Updating Homebrew...
Warning: libusbmuxd HEAD-1e7711d_1 is already installed and up-to-date
To reinstall HEAD_1, run `brew reinstall libusbmuxd`
Warning: Already linked: /usr/local/Cellar/libusbmuxd/HEAD-1e7711d_1
To relink: brew unlink libusbmuxd && brew link libusbmuxd
Updating Homebrew...
Warning: libimobiledevice HEAD-60823f9_4 is already installed and up-to-date
To reinstall HEAD_5, run `brew reinstall libimobiledevice`
Warning: Already linked: /usr/local/Cellar/libimobiledevice/HEAD-60823f9_4
To relink: brew unlink libimobiledevice && brew link libimobiledevice
Updating Homebrew...
Error: libplist 2.0.0_1 is already installed
To install HEAD, first run `brew unlink libplist`.
Warning: Already linked: /usr/local/Cellar/libplist/2.0.0_1
To relink: brew unlink libplist && brew link libplist
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
2019-11-14 15:24:02.355 system_profiler[84629:1595412] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:000002F4AE8BD514 IBFL:1C SRTG:[iBoot-1704.10]
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
2019-11-14 15:24:34.361 system_profiler[84982:1596630] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
2019-11-14 15:24:50.106 system_profiler[85325:1597384] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
2019-11-14 15:25:05.872 system_profiler[85789:1598286] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
2019-11-14 15:25:21.618 system_profiler[86133:1598994] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
[... 589 lines cut ...]
Extracting applelogo@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component AppleLogo...
Extracting batterycharging0@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component BatteryCharging0...
Extracting batterycharging1@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component BatteryCharging1...
Extracting batteryfull@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component BatteryFull...
Extracting batterylow0@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component BatteryLow0...
Extracting batterylow1@2x~iphone.s5l8960x.im4p...
Personalizing IMG4 component BatteryLow1...
Extracting glyphplugin@1136~iphone-lightning.s5l8960x.im4p...
Personalizing IMG4 component BatteryPlugin...
Extracting DeviceTree.n53ap.im4p...
Personalizing IMG4 component DeviceTree...
Extracting recoverymode@1136~iphone-lightning.s5l8960x.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Personalizing IMG4 component RestoreSEP...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
ERROR: Unable to receive message from FDR 0x7fcaf6dd7920 (-2). 0/2 bytes

gallus-x, 2019-11-14:
I have same problem, did you on high sierra?

Netfliics, 2019-11-14:
no I did it on Mojave

Merculous, 2019-11-18:
Futurerestore will fail no matter what during restore process sometimes. Just retry again. If everything is fine and working, please close this issue.

Merculous, 2019-11-18:
possible duplicate of #73