MatthewPierson/1033-OTA-Downgrader #73: ERROR: Unable to receive message from FDR 0x7fec32596ac0 (-2). 0/2 bytes After verifying restore -> checking filesystems

polishmathu, 2019-11-15:
It's iPhone 5 6,2 16GB and iOS 12.4.3 was installed. I ran the script on 15" rMBP 2015 with macOS 10.14.6 Mojave installed.  Everything seemed to go fine, the ipsw seemed to "install" and I got this error when checking filesystems (15) after verifying restore. 

[Terminal Saved Output .txt](https://github.com/MatthewPierson/1033-OTA-Downgrader/files/3853455/Terminal.Saved.Output.txt)

`[+] Installing dependencies
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
aliyun-cli      catch2          gatsby-cli      paket           terrahub
bro             exploitdb       git-annex       sonobuoy        tunnel

Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Error: openssl 1.0.2s is already installed
To upgrade to 1.0.2t, run `brew upgrade openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libusbmuxd HEAD-c2db27c_1 is already installed and up-to-date
To reinstall HEAD_1, run `brew reinstall libusbmuxd`
Warning: Already linked: /usr/local/Cellar/libusbmuxd/HEAD-c2db27c_1
To relink: brew unlink libusbmuxd && brew link libusbmuxd
Warning: libimobiledevice HEAD-60823f9_4 is already installed and up-to-date
To reinstall HEAD_5, run `brew reinstall libimobiledevice`
Warning: Already linked: /usr/local/Cellar/libimobiledevice/HEAD-60823f9_4
To relink: brew unlink libimobiledevice && brew link libimobiledevice
Error: libplist 2.0.0_1 is already installed
To install HEAD, first run `brew unlink libplist`.
Warning: Already linked: /usr/local/Cellar/libplist/2.0.0_1
To relink: brew unlink libplist && brew link libplist
Cloning into 'ipwndfu_public'...
remote: Enumerating objects: 463, done.
remote: Total 463 (delta 0), reused 0 (delta 0), pack-reused 463
Receiving objects: 100% (463/463), 1.87 MiB | 1.97 MiB/s, done.
Resolving deltas: 100% (239/239), done.
2019-11-15 23:37:42.835 system_profiler[38354:581073] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:00000476F1BB2DD8 IBFL:1C SRTG:[iBoot-1704.10]
Device is now in pwned DFU Mode.
(15.10 seconds)
2019-11-15 23:38:08.626 system_profiler[38694:581825] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:00000476F1BB2DD8 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Successfully applied patches
Resetting device state
* This will effectiveley disable pwned DFU Mode
* Only the signature patches will remain
Device is now ready to accept unsigned images
We seem to be in pwned DFU mode!
[+] Build folder does not exist! Grabbing dependencies and installing!
Cloning into 'libirecovery'...
remote: Enumerating objects: 65, done.
remote: Counting objects: 100% (65/65), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 1154 (delta 32), reused 50 (delta 20), pack-reused 1089
Receiving objects: 100% (1154/1154), 479.57 KiB | 442.00 KiB/s, done.
[... 2577 lines cut ...]
Done sending NORData
About to send RootTicket...
Sending RootTicket now...
Done sending RootTicket
Waiting for NAND (28)
Checking filesystems (15)
Checking filesystems (15)
Checking filesystems (15)
Unmounting filesystems (29)
Unmounting filesystems (29)
Creating partition map (11)
Creating filesystem (12)
About to send filesystem...
Connected to ASR
Validating the filesystem
Filesystem validated
Sending filesystem now...
[==================================================] 100.0%
Done sending filesystem
Verifying restore (14)
[==================================================] 100.0%
Checking filesystems (15)
ERROR: Unable to receive message from FDR 0x7fec32596ac0 (-2). 0/2 bytes

`

Merculous, 2019-11-18:
From what I've seen, this was just an error from idevicerestore/futurerestore that happens once in a while. Just re-run the script and try again. Please comment if there's any other issues present.

polishmathu, 2019-11-18:
Thank you for answering! I took a closer look at both of my Anker Lightning cables and on each of them there is a burned pin, maybe that's reason it didn't work. I'll get a new cable in a couple of days and let you know.