MatthewPierson/1033-OTA-Downgrader #74: error issue

shashoulqw, 2019-11-17:
Last login: Sun Nov 17 12:21:13 on ttys000
RCMs-MacBook-Pro:~ RCM$ /Users/RCM/Downloads/1033-OTA-Downgrader-master/restore.sh /Users/RCM/Downloads/1033-OTA-Downgrader-master/iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw 
[+] Installing dependencies
Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl 1.0.2t is already installed and up-to-date
To reinstall 1.0.2t, run `brew reinstall openssl`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: libusbmuxd HEAD-8f3afaf_1 is already installed and up-to-date
To reinstall HEAD_1, run `brew reinstall libusbmuxd`
Warning: Already linked: /usr/local/Cellar/libusbmuxd/HEAD-8f3afaf_1
To relink: brew unlink libusbmuxd && brew link libusbmuxd
Warning: libimobiledevice HEAD-af91dc6_5 is already installed and up-to-date
To reinstall HEAD_5, run `brew reinstall libimobiledevice`
Warning: Already linked: /usr/local/Cellar/libimobiledevice/HEAD-af91dc6_5
To relink: brew unlink libimobiledevice && brew link libimobiledevice
Error: libplist 2.1.0 is already installed
To install HEAD, first run `brew unlink libplist`.
==> `brew cleanup` has not been run in 30 days, running now...
Error: Permission denied @ apply2files - /usr/local/lib/docker/cli-plugins
Warning: Already linked: /usr/local/Cellar/libplist/2.1.0
To relink: brew unlink libplist && brew link libplist
From https://github.com/MatthewPierson/ipwndfu_public
 * branch            master     -> FETCH_HEAD
Already up to date.
2019-11-17 12:23:35.390 system_profiler[42507:71727230] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:00 ECID:0000000DCFD68F5C IBFL:1C SRTG:[iBoot-1704.10]
Device is now in pwned DFU Mode.
(13.19 seconds)
2019-11-17 12:23:59.738 system_profiler[43847:71731634] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:00 ECID:0000000DCFD68F5C IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Successfully applied patches
Resetting device state
* This will effectiveley disable pwned DFU Mode
* Only the signature patches will remain
Traceback (most recent call last):
  File "rmsigchks.py", line 114, in <module>
    main()
  File "rmsigchks.py", line 107, in main
    device.ctrl_transfer(HOST2DEVICE, DFU_ABORT, 0, 0, 0, 0)
  File "/Users/RCM/ipwndfu_public/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/Users/RCM/ipwndfu_public/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/Users/RCM/ipwndfu_public/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
We seem to be in pwned DFU mode!
[+] Build folder exists! If the script doesn't work please delete the 'Build' folder and run it again
ipsw/Firmware/aopfw.im4p
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware/dfu/iBEC.n69.RELEASE.im4p.plist
[... 208 lines cut ...]
   creating: ipsw/Firmware/usr/
   creating: ipsw/Firmware/usr/local/
   creating: ipsw/Firmware/usr/local/standalone/
  inflating: ipsw/kernelcache.release.iphone6  
  inflating: ipsw/kernelcache.release.iphone8b  
  inflating: ipsw/Restore.plist      
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw -> ./Mav7Mav8-7.60.00.Release.bbfw
Adlm				Pictures
Applications			Public
Desktop				StableA7
Documents			StableA7.sh
Downloads			build
Library				iCloud 云盘（归档）
Mav7Mav8-7.60.00.Release.bbfw	ipsw
Movies				ipwndfu_public
Music				shsh
/Users/RCM/Downloads/1033-OTA-Downgrader-master/restore.sh: line 108: ./igetnonce: No such file or directory
/Users/RCM/Downloads/1033-OTA-Downgrader-master/restore.sh: line 115: ./igetnonce: No such file or directory
/Users/RCM/Downloads/1033-OTA-Downgrader-master/restore.sh: line 122: ./igetnonce: No such file or directory
/Users/RCM/Downloads/1033-OTA-Downgrader-master/restore.sh: line 129: ./igetnonce: No such file or directory
/Users/RCM/Downloads/1033-OTA-Downgrader-master/restore.sh: line 136: ./igetnonce: No such file or directory
/Users/RCM/Downloads/1033-OTA-Downgrader-master/restore.sh: line 143: ./igetnonce: No such file or directory
Either unsupported device or no device found.
Exiting..
RCMs-MacBook-Pro:~ RCM$

MatthewPierson, 2019-11-17:
Download the latest version and try again. Delete all the old files first.

shashoulqw, 2019-11-17:
thanks i finally found i fogot first step cd the files :(

Merculous, 2019-11-18:
If everything went successful, please either close this issue or respond back with any other issues.

closed 2019-11-18 by shashoulqw