MatthewPierson/1033-OTA-Downgrader #78: pwned DFU mode stuck

iChernov, 2019-11-18:
Script works till entering the pwned DFU mode - after entering it the first log is:

Device is now in pwned DFU Mode.
(13.62 seconds)
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode

**and then:**

Device is already in pwned DFU Mode. Not executing exploit.
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode

and it runs the infinite loop of "waiting for DFU", while it is already in pwned DFU. Any ideas what to do?

MatthewPierson, 2019-11-18:
Can you please send the full log, can't do anything without it

jinjin123, 2019-12-27:
same,log `The script will run ipwndfu again and again until the device is in PWNDFU mode
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode`