MatthewPierson/1033-OTA-Downgrader #79: No Apple Device in DFU mode

aparajey, 2019-11-18:
Hello,

i am using MacOS Catalina and when i am running restore.sh i am getting following error. I am in DFU mode but script is not detecting Iphone.

2019-11-19 02:25:24.662 system_profiler[50127:1074590] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-19 02:25:24.671 system_profiler[50127:1074590] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.

Any suggestion?

MatthewPierson, 2019-11-18:
Try a different cable/USB port. Does iTunes detect the device in DFU mode?

aparajey, 2019-11-19:
yes iTunes detect Iphone in recovery/DFU mode. but script doesn't

qwqcode, 2019-11-20:
i have the same issue.
TAT

qwqcode, 2019-11-22:
I downgraded to mojave and it works well...

cs0663c, 2019-12-19:
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
ERROR: No Apple device in DFU Mode 0x1227 detected after 5.00 second timeout. Exiting.
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode

jinjin123, 2019-12-27:
> Attempting to get into pwndfu mode
> Please just enter DFU mode again on each reboot
> The script will run ipwndfu again and again until the device is in PWNDFU mode
> ERROR: No Apple device in DFU Mode [...]

same problem,I just downgraded my macpro2015 from mojave to high sierra,cannot test mojava ....

Stavi1010, 2020-07-03:
I am currently having the same issue but my iTunes is not detecting the device in dfu mode. What do I do?

lakiyo, 2020-09-24:
I have the same problem. macOS Catalina 10.15.6, iPad mini 2 

just infinity loop

zhangguorong1995, 2020-12-04:
I have the same problem. macOS Catalina 10.15.6, iPhone 5s

just infinity loop