MatthewPierson/1033-OTA-Downgrader #80: Python runing error

a1266143, 2019-11-19:
Python script Running error,
Mac os version:Hackintosh 10.15 catalina
phone:iphone5s(GSM)

MatthewPierson, 2019-11-19:
I need the full log to be able to help you

a1266143, 2019-11-20:
> I need the full log to be able to help you

thank you, I will upload log later

a1266143, 2019-11-20:
> I need the full log to be able to help you

this is part of the log

`Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:000001D4CE32F554 IBFL:1C SRTG:[iBoot-1704.10]
Traceback (most recent call last):
  File "./ipwndfu", line 66, in <module>
    checkm8.exploit()
  File "/Users/lee/Desktop/1033-OTA-Downgrader-master/ipwndfu_public/checkm8.py", line 533, in exploit
    libusb1_async_ctrl_transfer(device, 0x21, 1, 0, 0, 'A' * 0x800, 0.0001)
  File "/Users/lee/Desktop/1033-OTA-Downgrader-master/ipwndfu_public/checkm8.py", line 45, in libusb1_async_ctrl_transfer
    assert usb.backend.libusb1._lib.libusb_cancel_transfer(transfer_ptr) == 0
AssertionError
./restore.sh: line 311:  4378 Segmentation fault: 11  ./ipwndfu -p
No matching processes belonging to you were found
Waiting 10 seconds to allow you to enter DFU mode`

MatthewPierson, 2019-11-20:
I need the FULL log, not just part of it

a1266143, 2019-11-21:
> I need the FULL log, not just part of it

ok,thank you,i will upload full log later,thanks again

matthew-5pl, 2019-11-21:
Same here. Also on Catalina Hackintosh, 5s GSM, will send logs later

matthew-5pl, 2019-11-21:
Ok so here is mine:

matthew-5pl, 2019-11-21:
Process:               Python [15813]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.16 (2.7.16)
Build Info:            python-125000000000000~676
Code Type:             X86-64 (Native)
Parent Process:        bash [2023]
Responsible:           Terminal [1803]
User ID:               501

PlugIn Path:             /usr/local/Cellar/libusb/1.0.23/lib/libusb-1.0.0.dylib
PlugIn Identifier:       libusb-1.0.0.dylib
PlugIn Version:          ??? (0)

Date/Time:             2019-11-21 17:58:06.955 +0100
OS Version:            Mac OS X 10.15.1 (19B88)
Report Version:        12
Anonymous UUID:        41098E72-7F0E-4E35-A2DF-CF616491C7B7

Time Awake Since Boot: 3200 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [15813]

VM Regions Near 0x58:
--> 
    __TEXT                 000000010644d000-000000010644f000 [    8K] r-x/r-x SM=COW  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libusb-1.0.0.dylib            	0x0000000106a6c692 remove_from_flying_list + 17
1   libusb-1.0.0.dylib            	0x0000000106a6c7da usbi_handle_transfer_completion + 39
2   libusb-1.0.0.dylib            	0x0000000106a6d48f handle_events + 996
3   libusb-1.0.0.dylib            	0x0000000106a6ce4e libusb_handle_events_timeout_completed + 262
4   libusb-1.0.0.dylib            	0x0000000106a69927 libusb_exit + 221
5   libffi.dylib                  	0x00007fff712f4915 ffi_call_unix64 + 85
6   libffi.dylib                  	0x00007fff712f4255 ffi_call_int + 699
7   _ctypes.so                    	0x000000010681b650 _ctypes_callproc + 657
8   _ctypes.so                    	0x0000000106815b77 0x106813000 + 11127
9   org.python.python             	0x00007fff47218319 PyObject_Call + 97
10  org.python.python             	0x00007fff4729440a PyEval_EvalFrameEx + 20636
11  org.python.python             	0x00007fff47298a37 0x7fff4720e000 + 567863
12  org.python.python             	0x00007fff472942c2 PyEval_EvalFrameEx + 20308
13  org.python.python             	0x00007fff47298a37 0x7fff4720e000 + 567863
14  org.python.python             	0x00007fff472942c2 PyEval_EvalFrameEx + 20308
15  org.python.python             	0x00007fff47298a37 0x7fff4720e000 + 567863
16  org.python.python             	0x00007fff472942c2 PyEval_EvalFrameEx + 20308
17  org.python.python             	0x00007fff4728ed5e PyEval_EvalCodeEx + 534
18  org.python.python             	0x00007fff47235c85 0x7fff4720e000 + 162949
19  org.python.python             	0x00007fff47218319 PyObject_Call + 97
20  org.python.python             	0x00007fff47222afe 0x7fff4720e000 + 84734
21  org.python.python             	0x00007fff47218319 PyObject_Call + 97
22  org.python.python             	0x00007fff472983d9 PyEval_CallObjectWithKeywords + 159
23  org.python.python             	0x00007fff4725e7b7 0x7fff4720e000 + 329655
24  org.python.python             	0x00007fff47262ba2 0x7fff4720e000 + 347042
25  org.python.python             	0x00007fff4724307c 0x7fff4720e000 + 217212
26  org.python.python             	0x00007fff47262d34 0x7fff4720e000 + 347444
27  org.python.python             	0x00007fff4724307c 0x7fff4720e000 + 217212
28  org.python.python             	0x00007fff47262d34 0x7fff4720e000 + 347444
29  org.python.python             	0x00007fff47244356 0x7fff4720e000 + 222038
30  org.python.python             	0x00007fff47242c57 0x7fff4720e000 + 216151
31  org.python.python             	0x00007fff47242345 0x7fff4720e000 + 213829
[... 148 lines cut ...]
__UNICODE                          564K        1 
mapped file                         28K        1 
shared memory                       12K        3 
===========                     =======  ======= 
TOTAL                            519.8M      271 

Model: iMac19,1, BootROM 1037.40.118.0.0, 4 processors, Quad-Core Intel Core i5, 3,09 GHz, 4 GB, SMC 2.46f12
Graphics: NVIDIA GeForce GT 710, NVIDIA GeForce GT 710, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM3, 4 GB, DDR3, 1333 MHz, Kingston, 99U5584-005.A00LF
Bluetooth: Version 7.0.1f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
PCI Card: Intel 82579LM PCI Express Gigabit Ethernet, sppci_ethernet, Ethernet
PCI Card: NVIDIA GeForce GT 710, sppci_displaycontroller, PCI Slot 0
Serial ATA Device: KINGSTON SA400S37240G, 240,06 GB
Serial ATA Device: hp      DVD-RAM GH80N
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB2.0 WLAN
USB Device: Keyboard Hub
USB Device: Apple Keyboard
USB Device: BCM20702A0
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Apple Mobile Device (DFU Mode)
Thunderbolt Bus: