MatthewPierson/1033-OTA-Downgrader #82: Failed with errorcode=-11

YuriJJ16, 2019-11-19:
Help me please!
[LOG.txt](https://github.com/MatthewPierson/1033-OTA-Downgrader/files/3865939/LOG.txt)

fhc5326, 2019-11-23:
I got an Error,too!
Attach log for you..
[log2.txt](https://github.com/MatthewPierson/1033-OTA-Downgrader/files/3882504/log2.txt)

MatthewPierson, 2019-11-23:
@fhc5326 Delete the files you have now, download the latest version and use that. I don't know why you have a mixture of new and old files but you only need what is in the repo at the moment

fhc5326, 2019-11-24:
@MatthewPierson Thanks, I realize that because my openssl error so I failed(At libimobiledevice).
I recompile openssl and success, but actually I just want to reinstall some 32-bit apps only, but I'll upgrade back to 12.4.3 because some app issue.

YuriJJ16, 2019-11-26:
@MatthewPierson is there any errors in my log?

MatthewPierson, 2019-11-28:
I don’t know what’s wrong for you. I’d just redownload everything and try again