MatthewPierson/1033-OTA-Downgrader #85: dyld: Library not loaded: /usr/local/opt/usbmuxd/lib/libusbmuxd.4.dylib   Referenced from: /rsu/local/lib/libimobiledevice.6.dylib   Reason: image not found

Amaise777666, 2019-11-23:
Last login: Sat Nov 23 01:12:19 on console
iMac-Ama:~ ama$ bash <(curl -s https://gitlab.com/snippets/1907816/raw)
==> If you want to run StableA7 again, use this command: ./StableA7.sh
==> Press enter/return to run StableA7... 
==> Checking for brew...
==> Installing dependencies...
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> New Formulae
cargo-c             devdash             monolith            postgresql@11
==> Updated Formulae
citus               libpq               pgrouting           wal2json
cstore_fdw          libpqxx             postgis
eslint              pgloader            postgresql
fluxctl             pgroonga            temporal_tables

Warning: autoconf 2.69 is already installed and up-to-date
To reinstall 2.69, run `brew reinstall autoconf`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: bsdiff 4.3 is already installed and up-to-date
To reinstall 4.3, run `brew reinstall bsdiff`
Warning: libimobiledevice 1.2.0_6 is already installed and up-to-date
To reinstall 1.2.0_6, run `brew reinstall libimobiledevice`
Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl@1.1 1.1.1d is already installed and up-to-date
To reinstall 1.1.1d, run `brew reinstall openssl@1.1`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Warning: wget 1.20.3_2 is already installed and up-to-date
To reinstall 1.20.3_2, run `brew reinstall wget`
==> Checking for libirecovery...
==> Checking for libfragmentzip...

**************** StableA7 ****************

=> By Luke:
=>   - u/TheLukeGuy
=>   - @ConsoleLogLuke

==> WARNING: This is a beta version. Things might not work properly or at all.

==> Which futurerestore patch would you like to use?
==> If you are unsure, choose option 1. This is only present in this beta version.

=> (1) Normal/Auto
=>   - Attempts to perform a restore without custom dylibs
=>   - Inconsistent, sometimes works and sometimes doesn't
=>   - Works on macOS High Sierra, Mojave, and Catalina
=>   - Doesn't require root
=>   - Switches to rsu patch if it fails

=> (2) /rsu
=>   - Makes futurerestore look for custom dylibs in /rsu instead of /usr
=>   - Only works on macOS High Sierra and Mojave
=>   - Works on macOS Catalina with SIP disabled and / remounted as rw
=>   - Requires root

==> Which patch would you like to use? (1/2): 2
==> Creating work directory...
==> Downloading binaries...
Warning: Failed to set locale category LC_NUMERIC to ru_BY.
Warning: Failed to set locale category LC_TIME to ru_BY.
Warning: Failed to set locale category LC_COLLATE to ru_BY.
Warning: Failed to set locale category LC_MONETARY to ru_BY.
Warning: Failed to set locale category LC_MESSAGES to ru_BY.
bin.zip             100%[===================>]   2.05M  1.41MB/s    за 1.4s    
==> Downloading patches...
Warning: Failed to set locale category LC_NUMERIC to ru_BY.
Warning: Failed to set locale category LC_TIME to ru_BY.
Warning: Failed to set locale category LC_COLLATE to ru_BY.
Warning: Failed to set locale category LC_MONETARY to ru_BY.
Warning: Failed to set locale category LC_MESSAGES to ru_BY.
patch.zip           100%[===================>]   1.19M   953KB/s    за 1.3s    
==> Renaming futurerestore binary...
==> Downloading rsu...
Warning: Failed to set locale category LC_NUMERIC to ru_BY.
Warning: Failed to set locale category LC_TIME to ru_BY.
Warning: Failed to set locale category LC_COLLATE to ru_BY.
Warning: Failed to set locale category LC_MONETARY to ru_BY.
Warning: Failed to set locale category LC_MESSAGES to ru_BY.
rsu.zip             100%[===================>]   1.43M  1.11MB/s    за 1.3s    
==> Moving rsu to /rsu. This might ask for your password...
Password:
==> Waiting for device...
==> iPhone6,1 found!
==> Your iPhone 5S is supported!

==> If you already downloaded an IPSW file, you can use it with one of these two options:

=>   - Move the IPSW file to the StableA7 folder and press enter/return
=>   - Drag the IPSW file into this Terminal window and press enter/return

==> If you would like the script to download the IPSW file, just press enter/return... /Users/ama/Desktop/iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw 
==> Copying IPSW...
==> Checking for IPSW...
==> IPSW found!
==> Extracting IPSW...
==> Copying iBEC/iBSS...
==> Patching iBEC/iBSS...
==> Copying patched iBEC/iBSS to IPSW...
==> Creating custom IPSW...
==> Cleaning up...
==> Downloading ipwndfu...
Warning: Failed to set locale category LC_NUMERIC to ru_BY.
Warning: Failed to set locale category LC_TIME to ru_BY.
Warning: Failed to set locale category LC_COLLATE to ru_BY.
Warning: Failed to set locale category LC_MONETARY to ru_BY.
Warning: Failed to set locale category LC_MESSAGES to ru_BY.
ipwndfu.zip             [     <=>            ]   1.31M  1.09MB/s    за 1.2s    
==> Please put your device in DFU mode and press enter/return... 
==> Closing all iTunes/Finder windows...
==> Attempting to enter pwned DFU mode...
==> Exploit failed. Please exit DFU mode.
==> Please put your device in DFU mode and press enter/return... 
==> Closing all iTunes/Finder windows...
==> Attempting to enter pwned DFU mode...
==> Removing signature checks...
==> Sending test file to device...
[==================================================] 100.0%
==> Sending patched iBSS/iBEC to device...
[==================================================] 100.0%
[==================================================] 100.0%
==> Waiting for device to reconnect...
==> Downloading OTA manifests...
Warning: Failed to set locale category LC_NUMERIC to ru_BY.
Warning: Failed to set locale category LC_TIME to ru_BY.
Warning: Failed to set locale category LC_COLLATE to ru_BY.
Warning: Failed to set locale category LC_MONETARY to ru_BY.
Warning: Failed to set locale category LC_MESSAGES to ru_BY.
manifests.zip       100%[===================>]  24.64K   163KB/s    за 0.2s    
==> Getting ECID and ApNonce...
==> Copying OTA manifest...
==> Copying baseband...
==> Copying SEP...
==> Requesting ticket...
==> Cleaning up...

==> Preparation complete! Press enter/return to restore... 
==> Restoring device to 10.3.3...
dyld: Library not loaded: /usr/local/opt/usbmuxd/lib/libusbmuxd.4.dylib
  Referenced from: /rsu/local/lib/libimobiledevice.6.dylib
  Reason: image not found
./StableA7.sh: line 352: 52374 Abort trap: 6           bin/futurerestore -t ota.shsh -s sep.im4p -m BuildManifest.plist -b baseband.bbfw -p BuildManifest.plist custom.ipsw

==> Restoring failed. Attempting to exit recovery mode...
./StableA7.sh: line 352: 52376 Abort trap: 6           bin/futurerestore --exit-recovery >&/dev/null
iMac-Ama:~ ama$

MatthewPierson, 2019-11-23:
That is not my/@Merculous  script. I'm not gonna give help for something I didn't make

closed 2019-11-23 by MatthewPierson

Amaise777666, 2019-11-23:
Если не сложно пришли ссылку на свой скрипт. Проверю и его

С уважением,
Карачун Денис Александрович
+375 33 323 26 70
karachund@list.ru