MatthewPierson/1033-OTA-Downgrader #86: ./restore.sh: line 292: futurerestore: command not found (iPhone 5s, Mojave 10.14.6, script is last version)

nafign1ck, 2019-11-23:
Last login: Sat Nov 23 18:37:33 on ttys000
Neulovimiy:~ neulovimiy$ ls
Desktop		Downloads	Movies		Pictures	StableA7
Documents	Library		Music		Public		StableA7.sh
Neulovimiy:~ neulovimiy$ cd Desktop/
Neulovimiy:Desktop neulovimiy$ ls
1033-OTA-Downgrader-master
Снимок экрана 2019-11-23 в 16.47.20.png
Запись экрана 2019-11-23 в 16.47.31.mov
Рабочий стол — Neulovimiy
Neulovimiy:Desktop neulovimiy$ cd 1033-OTA-Downgrader-master/
Neulovimiy:1033-OTA-Downgrader-master neulovimiy$ ls
README.md
iBoot64Patcher
iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
igetnonce
manifests
restore.sh
Neulovimiy:1033-OTA-Downgrader-master neulovimiy$ ./restore.sh iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw 
[+] Installing dependencies
Warning: libtool 2.4.6_1 is already installed and up-to-date
To reinstall 2.4.6_1, run `brew reinstall libtool`
Warning: automake 1.16.1_1 is already installed and up-to-date
To reinstall 1.16.1_1, run `brew reinstall automake`
Warning: lsusb 1.0 is already installed and up-to-date
To reinstall 1.0, run `brew reinstall lsusb`
Warning: openssl@1.1 1.1.1d is already installed and up-to-date
To reinstall 1.1.1d, run `brew reinstall openssl@1.1`
Warning: libzip 1.5.2 is already installed and up-to-date
To reinstall 1.5.2, run `brew reinstall libzip`
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
Error: libusbmuxd 2.0.0_1 is already installed
To install HEAD_1, first run `brew unlink libusbmuxd`.
Warning: Skipping (old) /usr/local/Cellar/libusbmuxd/2.0.0_1 due to it being linked
Warning: Already linked: /usr/local/Cellar/libusbmuxd/2.0.0_1
To relink: brew unlink libusbmuxd && brew link libusbmuxd
Error: libimobiledevice 1.2.0_6 is already installed
To install HEAD_6, first run `brew unlink libimobiledevice`.
Warning: Skipping (old) /usr/local/Cellar/libimobiledevice/1.2.0_6 due to it being linked
Warning: Already linked: /usr/local/Cellar/libimobiledevice/1.2.0_6
To relink: brew unlink libimobiledevice && brew link libimobiledevice
Error: libplist 2.1.0 is already installed
To install HEAD, first run `brew unlink libplist`.
Warning: Skipping (old) /usr/local/Cellar/libplist/2.1.0 due to it being linked
Warning: Already linked: /usr/local/Cellar/libplist/2.1.0
To relink: brew unlink libplist && brew link libplist
Cloning into 'ipwndfu_public'...
remote: Enumerating objects: 463, done.
remote: Total 463 (delta 0), reused 0 (delta 0), pack-reused 463
Receiving objects: 100% (463/463), 1.87 MiB | 164.00 KiB/s, done.
Resolving deltas: 100% (239/239), done.
2019-11-23 19:26:40.753 system_profiler[10779:59065] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Waiting 10 seconds to allow you to enter DFU mode
Attempting to get into pwndfu mode
Please just enter DFU mode again on each reboot
The script will run ipwndfu again and again until the device is in PWNDFU mode
*** checkm8 exploit by axi0mX ***
*** modified version by Linus Henze ***
*** s5l8965x support by Matthew Pierson ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:000007372A7680D0 IBFL:1C SRTG:[iBoot-1704.10]
Device is now in pwned DFU Mode.
(15.60 seconds)
2019-11-23 19:27:07.577 system_profiler[11167:60290] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:000007372A7680D0 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Successfully applied patches
Resetting device state
* This will effectiveley disable pwned DFU Mode
[... 1932 lines cut ...]
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/LLB.n69u.RELEASE.im4p
ipsw/Firmware/all_flash
ipsw/Firmware/Mav10-5.62.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav10-5.62.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/kernelcache.release.iphone8b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/kernelcache.release.iphone6
ipsw/BuildManifest.plist
ipsw/058-74917-062.dmg
ipsw/._058-74917-062.dmg
ipsw
downgrade.ipsw
iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
Neulovimiy:1033-OTA-Downgrader-master neulovimiy$ brew search futurestore
No formula or cask found for "futurestore".
Neulovimiy:1033-OTA-Downgrader-master neulovimiy$

nafign1ck, 2019-11-23:
Also why is identified as iPhone6,2?

MatthewPierson, 2019-11-23:
Delete all the files and try again. Also "iPhone6,2" is the model identifier for the Global iPhone 5s

nafign1ck, 2019-11-24:
I deleted all files including iOS 10.3.3, downloaded and try again, still unchanged:
Version: 78c1203bcb326dcc813cc24cce2df6b88c01eef4 - 301
libfragmentzip version: 0.58-ffec726ea514d8b8f3f3e42d04ce1430f3889747
[TSSC] manually specified ECID to use, parsed "7933517005008" to dec:7933517005008 hex:7372a7680d0
[TSSC] manually specified apnonce to use, parsed "d1c243d4e68964a383fe9804dc63d6a359cca4b6" to hex:d1c243d4e68964a383fe9804dc63d6a359cca4b6
[TSSC] opening manifests/BuildManifest_iPhone6,2_1033_OTA.plist
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... success
also requesting APTicket for installType=Update
[Error] [TSSR] Error: could not get id0 for installType=Update
[WARNING] [TSSR] failed to build tssrequest for alternative installType
[TSSR] User specified doesn't to request a baseband ticket.
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved shsh blobs!

iOS 10.3.3 for device iPhone6,2 IS being signed!
7933517005008_iPhone6,2_10.3.3-14G60_d1c243d4e68964a383fe9804dc63d6a359cca4b6.shsh -> shsh/apnonce.shsh2
Done prepping files! Time to downgrade!!!
****RESTORING!****
Waiting for device to reconnect...
./restore.sh: line 292: futurerestore: command not found
Cleaning up :D
dummy_file
iBSS.im4p
iBSS.img4
iBSS.iphone6.RELEASE.im4p
iBSS.prepatched
iBSS.raw
iBEC.im4p
iBEC.img4
iBEC.iphone6.RELEASE.im4p
iBEC.prepatched
iBEC.raw
Mav7Mav8-7.60.00.Release.bbfw
sep-firmware.n53.RELEASE.im4p
ipsw/Firmware/aopfw.im4p
ipsw/Firmware/usr/local/standalone
ipsw/Firmware/usr/local
ipsw/Firmware/usr
ipsw/Firmware/dfu/iBEC.n69.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.n69u.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.iphone6.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.n69.RELEASE.im4p
ipsw/Firmware/dfu/iBEC.n69u.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.n69u.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.iphone6.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.iphone6.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.n69.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBEC.iphone6.RELEASE.im4p.plist
ipsw/Firmware/dfu/iBSS.n69u.RELEASE.im4p
ipsw/Firmware/dfu/iBSS.n69.RELEASE.im4p
ipsw/Firmware/dfu
ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p.plist
ipsw/Firmware/all_flash/batterycharging0@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s8000.im4p.plist
ipsw/Firmware/all_flash/sep-firmware.n51.RELEASE.im4p
ipsw/Firmware/all_flash/iBoot.n69u.RELEASE.im4p
ipsw/Firmware/all_flash/sep-firmware.n51.RELEASE.im4p.plist
ipsw/Firmware/all_flash/LLB.n69.RELEASE.im4p
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s5l8960x.im4p
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8003.im4p
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/recoverymode@1136~iphone-lightning.s5l8960x.im4p.plist
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s8000.im4p
ipsw/Firmware/all_flash/sep-firmware.n69u.RELEASE.im4p.plist
ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s5l8960x.im4p.plist
[... 54 lines cut ...]
ipsw/Firmware/all_flash/sep-firmware.n53.RELEASE.im4p
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/LLB.n69u.RELEASE.im4p
ipsw/Firmware/all_flash
ipsw/Firmware/Mav10-5.62.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav10-5.62.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/kernelcache.release.iphone8b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/kernelcache.release.iphone6
ipsw/BuildManifest.plist
ipsw/058-74917-062.dmg
ipsw/._058-74917-062.dmg
ipsw
downgrade.ipsw
iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

colorzpe, 2019-12-11:
same problem

gblgbl1234, 2020-01-05:
is ok now ？

MatthewPierson, 2020-01-05:
Don't use this script anymore, use [Vieux](https://github.com/MatthewPierson/Vieux)

closed 2020-01-05 by MatthewPierson