MatthewPierson/1033-OTA-Downgrader #87: I do not understand what is happening (

Amaise777666, 2019-11-23:
[log1.TXT](https://github.com/MatthewPierson/1033-OTA-Downgrader/files/3882975/log1.TXT)
[Log2.txt](https://github.com/MatthewPierson/1033-OTA-Downgrader/files/3882976/Log2.txt)
[log3.TXT](https://github.com/MatthewPierson/1033-OTA-Downgrader/files/3882977/log3.TXT)

MatthewPierson, 2019-11-23:
It looks like your device is taking too long to reconnect when putting it into pwnrec mode. Trying increasing the value for "sleep" on line 276, 278 and 280 to something like 10 and see if that fixes it. You can open the script in a text editor and make those changes

Amaise777666, 2019-11-23:
thanks, I’ll try tomorrow, and I will write whether it helped or not

Amaise777666, 2019-11-24:
Did as you said.
[log 5.TXT](https://github.com/MatthewPierson/1033-OTA-Downgrader/files/3883525/log.5.TXT)
Somewhere in this place the device came out of pwn DFU
[dfu diskonect 1.TXT](https://github.com/MatthewPierson/1033-OTA-Downgrader/files/3883529/dfu.diskonect.1.TXT)

LifeIsBinary, 2019-12-31:
@Amaise777666  is your problem sloved? I met the same things in my iPad mini2 downgrading.