MatthewPierson/1033-OTA-Downgrader #88: I have a lot of issue.

FightMan01, 2019-11-24:
Hi. I'm trying to use this tool, but I'm having a lot of issue.
Here is the section where the issues are located:
```
configure: error: Package requirements (libplist >= 1.0) were not met:

No package 'libplist' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libplist_CFLAGS
and libplist_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
make: *** No targets specified and no makefile found.  Stop.
[+] Dependencies should now be installed and compiled.
shsh
Killing iTunes as this will be quite annoying with what we are going to do.
Archive:  iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
  inflating: ipsw/058-74917-062.dmg  
  inflating: ipsw/._058-74917-062.dmg  
  inflating: ipsw/058-74940-063.dmg  
  inflating: ipsw/058-75381-062.dmg  
  inflating: ipsw/BuildManifest.plist  
   creating: ipsw/Firmware/
   creating: ipsw/Firmware/all_flash/
  inflating: ipsw/Firmware/all_flash/applelogo@2x~iphone.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/applelogo@2x~iphone.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/applelogo@2x~iphone.s8000.im4p  
  inflating: ipsw/Firmware/all_flash/applelogo@2x~iphone.s8000.im4p.plist  
  inflating: ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p  
  inflating: ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~iphone.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~iphone.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~iphone.s8000.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~iphone.s8000.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~iphone.s8003.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging0@2x~iphone.s8003.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~iphone.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~iphone.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~iphone.s8000.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~iphone.s8000.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~iphone.s8003.im4p  
  inflating: ipsw/Firmware/all_flash/batterycharging1@2x~iphone.s8003.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~iphone.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~iphone.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8003.im4p  
  inflating: ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8003.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterylow0@2x~iphone.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterylow0@2x~iphone.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8000.im4p  
  inflating: ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8000.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p  
  inflating: ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterylow1@2x~iphone.s5l8960x.im4p  
  inflating: ipsw/Firmware/all_flash/batterylow1@2x~iphone.s5l8960x.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterylow1@2x~iphone.s8000.im4p  
  inflating: ipsw/Firmware/all_flash/batterylow1@2x~iphone.s8000.im4p.plist  
  inflating: ipsw/Firmware/all_flash/batterylow1@2x~iphone.s8003.im4p  
  inflating: ipsw/Firmware/all_flash/batterylow1@2x~iphone.s8003.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.n53ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.n53ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.n69ap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.n69ap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/DeviceTree.n69uap.im4p  
  inflating: ipsw/Firmware/all_flash/DeviceTree.n69uap.im4p.plist  
  inflating: ipsw/Firmware/all_flash/glyphplugin@1136~iphone-lightning.s5l8960x.im4p  
[... 296 lines cut ...]
ipsw/Firmware/all_flash/batteryfull@2x~iphone.s8000.im4p.plist
ipsw/Firmware/all_flash/DeviceTree.n51ap.im4p
ipsw/Firmware/all_flash/batterylow0@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/applelogo@2x~iphone.s8003.im4p.plist
ipsw/Firmware/all_flash/LLB.n69u.RELEASE.im4p
ipsw/Firmware/all_flash
ipsw/Firmware/Mav10-5.62.00.Release.plist
ipsw/Firmware/Mav7Mav8-7.60.00.Release.plist
ipsw/Firmware/Mav10-5.62.00.Release.bbfw
ipsw/Firmware/Mav7Mav8-7.60.00.Release.bbfw
ipsw/Firmware
ipsw/058-75381-062.dmg
ipsw/kernelcache.release.iphone8b
ipsw/058-74940-063.dmg
ipsw/Restore.plist
ipsw/kernelcache.release.iphone6
ipsw/BuildManifest.plist
ipsw/058-74917-062.dmg
ipsw/._058-74917-062.dmg
ipsw
downgrade.ipsw
iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D
```

FightMan01, 2019-11-24:
[image]

Ohh and this.

Zyzbs, 2019-12-12:
./restore.sh: line 177: img4tool: command not found
./restore.sh: line 178: img4tool: command not found
main: Starting...
libc++abi.dylib: terminating with uncaught exception of type tihmstar::OFexception: std::exception
./restore.sh: line 311: 71310 Abort trap: 6           ./iBoot64Patcher iBSS.raw iBSS.prepatched
main: Starting...
libc++abi.dylib: terminating with uncaught exception of type tihmstar::OFexception: std::exception
./restore.sh: line 311: 71311 Abort trap: 6           ./iBoot64Patcher iBEC.raw iBEC.prepatched
./restore.sh: line 181: img4tool: command not found
./restore.sh: line 182: img4tool: command not found
./restore.sh: line 183: tsschecker: command not found
mv: rename shsh/*.shsh* to shsh/stitch.shsh2: No such file or directory
./restore.sh: line 185: img4tool: command not found
./restore.sh: line 186: img4tool: command not found
cp: iBSS.img4: No such file or directory
cp: iBEC.img4: No such file or directory

### You need to download these through the script, these tools didn't installed, so it says No such file or directory

beryanow, 2020-01-26:
Solved?

beryanow, 2020-01-26:
Same issue