MatthewPierson/1033-OTA-Downgrader #91: [Error] failed to read SEP [Error] Fail code=-15 Failed with errorcode=-15

cryptlord4, 2019-11-30:
need help .... T-T....

OS : catalina 10.15.1
iPhone6,2 (iPhone 5S)

full log--------------------------------------------------------------------------------------------
*** SecureROM Signature check remover by Linus Henze ***
Found: CPID:8960 CPRV:11 CPFM:03 SCEP:01 BDID:02 ECID:00000077F4411794 IBFL:1C SRTG:[iBoot-1704.10] PWND:[checkm8]
Applying patches...
Successfully applied patches
Resetting device state
* This will effectiveley disable pwned DFU Mode
* Only the signature patches will remain
Device is now ready to accept unsigned images
We seem to be in pwned DFU mode!
[+] Build folder does not exist! Grabbing dependencies and installing!
Cloning into 'libirecovery'...
remote: Enumerating objects: 69, done.
remote: Counting objects: 100% (69/69), done.
remote: Compressing objects: 100% (49/49), done.
remote: Total 1158 (delta 35), reused 51 (delta 20), pack-reused 1089
Receiving objects: 100% (1158/1158), 479.92 KiB | 697.00 KiB/s, done.
Resolving deltas: 100% (567/567), done.
Cloning into 'libfragmentzip'...
remote: Enumerating objects: 79, done.
remote: Counting objects: 100% (79/79), done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 385 (delta 36), reused 64 (delta 25), pack-reused 306
Receiving objects: 100% (385/385), 288.33 KiB | 617.00 KiB/s, done.
Resolving deltas: 100% (154/154), done.
Cloning into 'libgeneral'...
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 137 (delta 0), reused 4 (delta 0), pack-reused 132
Receiving objects: 100% (137/137), 19.66 KiB | 3.93 MiB/s, done.
Resolving deltas: 100% (32/32), done.
Cloning into 'futurerestore'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 1070 (delta 0), reused 1 (delta 0), pack-reused 1066
Receiving objects: 100% (1070/1070), 263.61 KiB | 576.00 KiB/s, done.
Resolving deltas: 100% (652/652), done.
Submodule 'external/idevicerestore' (https://github.com/s0uthwest/idevicerestore.git) registered for path 'external/idevicerestore'
Submodule 'external/img4tool' (https://github.com/s0uthwest/img4tool.git) registered for path 'external/img4tool'
Submodule 'external/tsschecker' (https://github.com/s0uthwest/tsschecker.git) registered for path 'external/tsschecker'
Cloning into '/Users/seokcheol/Downloads/1033-OTA-Downgrader-master/build/futurerestore/external/idevicerestore'...
remote: Enumerating objects: 3287, done.        
remote: Total 3287 (delta 0), reused 0 (delta 0), pack-reused 3287        
Receiving objects: 100% (3287/3287), 1.05 MiB | 1.05 MiB/s, done.
Resolving deltas: 100% (2488/2488), done.
Cloning into '/Users/seokcheol/Downloads/1033-OTA-Downgrader-master/build/futurerestore/external/img4tool'...
remote: Enumerating objects: 624, done.        
remote: Total 624 (delta 0), reused 0 (delta 0), pack-reused 624        
Receiving objects: 100% (624/624), 157.94 KiB | 382.00 KiB/s, done.
Resolving deltas: 100% (341/341), done.
Cloning into '/Users/seokcheol/Downloads/1033-OTA-Downgrader-master/build/futurerestore/external/tsschecker'...
remote: Enumerating objects: 1943, done.        
remote: Total 1943 (delta 0), reused 0 (delta 0), pack-reused 1943        
Receiving objects: 100% (1943/1943), 873.22 KiB | 1.04 MiB/s, done.
Resolving deltas: 100% (1119/1119), done.
Submodule path 'external/idevicerestore': checked out 'b10634c81b47bc6842538358e889063e9cbcdbd4'
Submodule path 'external/img4tool': checked out 'dd9b385e6954b8d12021fd0c56e53d5ba760b80a'
Submodule path 'external/tsschecker': checked out 'c0b554e83f54d39d90cac9791160bf2ccb062aed'
Submodule 'external/jssy' (https://github.com/tihmstar/jssy.git) registered for path 'external/tsschecker/external/jssy'
Cloning into '/Users/seokcheol/Downloads/1033-OTA-Downgrader-master/build/futurerestore/external/tsschecker/external/jssy'...
remote: Enumerating objects: 117, done.        
remote: Total 117 (delta 0), reused 0 (delta 0), pack-reused 117        
Receiving objects: 100% (117/117), 25.80 KiB | 6.45 MiB/s, done.
[... 2221 lines cut ...]
[TSSR] Request URL set to https://gs.apple.com/TSS/controller?action=2
[TSSR] Sending TSS request attempt 1... failure
[Error] ERROR: TSS request failed (status=128, message=An internal error occurred.)
Saved shsh blobs!

iOS 10.3.3 for device iPhone6,2 IS being signed!
515199014804_iPhone6,2_10.3.3-14G60_be6cd33cfe780486f604015abb9c0910c2a5f6c3.shsh -> shsh/apnonce.shsh2
Done prepping files! Time to downgrade!!!
****RESTORING!****
Waiting for device to reconnect...
Version: 7e66824796f697ea54597dadbdba462382f004b9 - 247
Odysseus support: no
[INFO] 64-bit device detected
futurerestore init done
reading signing ticket shsh/apnonce.shsh2 is done
Found device iPhone6,2 n53ap
[Error] failed to read SEP
[Error] Fail code=-15
Failed with errorcode=-15
Cleaning up :D
dummy_file
ipsw
iPhone_4.0_64bit_10.3.3_14G60_Restore.ipsw
If you see this, we're done! Shoutout to the devs and Matty for making this possible! - Merculous
P.S. You know, this could look even better and be even easier if we port it to Python :D

blackCr0wn92, 2019-12-01:
Have the same issue. Maybe the problem lies with futurerestore and has to be fixed by tihmstar. I have tried all possible paths but no one came to success

cryptlord4, 2019-12-02:
> Have the same issue. Maybe the problem lies with futurerestore and has to be fixed by tihmstar. I have tried all possible paths but no one came to success

Nooooooooooooooooooooooooooooooooo

cryptlord4, 2019-12-13:
> Have the same issue. Maybe the problem lies with futurerestore and has to be fixed by tihmstar. I have tried all possible paths but no one came to success

Did you solve the problem?

blackCr0wn92, 2019-12-13:
No, but yesterday Matthew release the new tool vieux. I will try it on Sunday and report whether it worked

cryptlord4, 2019-12-13:
> No, but yesterday Matthew release the new tool vieux. I will try it on Sunday and report whether it worked

oH Thx!! bro